home-assistant/hassbian-scripts #12: Add hassbian-config status

balloob, 2017-02-08:
For integration into the frontend, we need a `hassbian-config status` command. The command will return (preferably in JSON) for each script the name, description and status (not_installed, installing, installed, est install time?).

Requires https://github.com/home-assistant/hassbian-scripts/pull/11 to be merged

**Use case:**
So I am planning on adding a hassbian panel to the Home Assistant UI to trigger these scripts. I am thinking about the following implementation:

 - whitelist the hass user sudo rights to call `hassbian-config`
 - hassbian component will call `hassbian-config status` and expose it as an API
 - hassbian panel in UI will show status of the different scripts
 - Future: see wifi settings etc etc

kxtcd950, 2017-02-08:
Obviously, this is only software.  Shell scripting is really very powerful, and I can't see any reason why it can't be coerced into implementing status on top of what we've got already with #11 under consideration.
The hassbian image has got /usr/bin/flock installed by default; this makes a package status system do-able; the database (actually a text file) can be locked against modifications made by other instances of the script.

Once #11 has been merged (I'm assuming at this point that it will be), we'd need to add additional functions to the ```install_*.sh``` scripts to provide the new information (name and status; description is already there), and add a new command to hassbian-config to collect the information in the format of your choice.

How do you see the python frontend collecting the JSON from the ```hassbian-config``` script?  grabbing stdout, a dedicated pipe (might be handy for real-time progress reports) or some other mechanism?
The only thing I'd suggest is that grabbing stdout - whilst easy - is subject to output buffering, which means that real-time progress won't work as well (workarounds do exist though).

I can't see a problem making JSON in the shell script; it'd be just hardcoded structure in text form (anathema to python people, I know, but bash doesn't "do" JSON natively).

Something else I've been thinking about (which may be a step too far, I realise) is that hassbian runs on a base Debian system.  Debian has a really good package manager already in ```apt-get```.  Subject to the homeassistant project being able to host a .deb repository and the participants gaining the correct knowledge to be able to author debian packages, the whole package management issue becomes as simple (!) as using apt and it's existing package management infrastructure.  Less software written more or less equates to fewer bugs in the system.  Whilst scripts would need to be written for configuration of the packages, the scripts would be contained within the .deb files, and it would be ```apt-get```'s job to make sure they're called at the correct time and in the correct manner.

balloob, 2017-02-08:
I was thinking about a more basic approach for the integration with `hassbian-config` (mainly because I don't know how to shell): just calling the status every 5 seconds and have the frontend poll that.

I agree that shell scripting is quite powerful. I'm actually quite impressed in how well you have been organizing it. A true artist 😉 

APT sources would indeed be the best. I think that for libopenzwave the next Debian should have us covered at least.

kxtcd950, 2017-02-09:
Calling the status every 5 seconds seems a little overkill for an operation which shouldn't be performed that often.
Is there a possibility that the ```hassbian-config install <suite>``` command could somehow prod homeassistant (via a websocket call?) to tell homeassistant that the installer is doing something, at which point homeassistant polls every 5 seconds, and upon receiving a "done" signal stops polling?  This would mean less overhead for the computer, and does fit better into the low-power (hue polling for tap devices, ugh!) modes of operation that the pi implies...
Given that hassbian-config is (should be, at least) the only way that the states of the packages change, having the notification of install operation start to homeassistant for beginning it's polling period seems a little more elegant.

balloob, 2017-02-09:
Yes, I would only poll when kicked off from the frontend and only as long as the Hassbian config panel is open.

kxtcd950, 2017-02-10:
Ok, good.  But that doesn't pick up the situation where a user (like us, quite familiar and comfortable in the command line) runs the hassbian-config manually to install a suite.
The UI is then left outdated, and the user possibly confused as to the state of the package they just installed?
Is there any external way of firing events on the internal service bus? By which I mean the same way that the "Services" UI component fires a Domain/Service/Data item?  If we could poke that from the outside, the shell script could notify the front-end via that mechanism that software is being installed.

balloob, 2017-02-10:
As long as a user has the page open that shows the status of packages, it will poll the status every 5 seconds. The only way to do external probing right now is via the API but that might be authenticated.

balloob, 2017-02-11:
This is what I have for now. Still need to hook it up to the real tool.

[image]

Landrash, 2017-02-11:
OMG. Your penguin is so cute!
If you two feel the tool output is done I'll try to create a image for you to test it out?

balloob, 2017-02-12:
This was the old version, I have since migrated the hassbian panel into a configuration panel with different sections. So no more penguin

Landrash, 2017-02-25:
@kxtcd950 
Trying out your changes but having a problem
That's what I get when running the command in question.
It's in a new location but I've updated the path variables so can't figure out why it errors out
```shell
pi@hassbian:/tmp $ hassbian-config show
List of suites available for installation:
/usr/local/bin/hassbian-config: line 19: /opt/hassbian/suites/install_get-all-suite-installers.sh: No such file or directory
/usr/local/bin/hassbian-config: line 20: get-all-suite-installers-show-long-info: command not found
get-all-suite-installers:
```

After this bug is fixed this feature is done :)

kxtcd950, 2017-02-25:
That looks like the hassbian-config script isn't being able to source the individual scripts to add the functions into the namespace; the mixture of _ and - characters in the line 19 error description looks screwy - I used _ in filenames, and - in function names exclusively.
That error referencing line 19 ( file /opt/hassbian/suites/install_get-all-suite-installers.sh) seems like it's not managed to find the script correctly; more accurately that the file glob which captures all the script names hasn't been capturing the correct thing and the function name/filename builder has malfunctioned as a result.
Any chance I can get hold of the image you're testing, so I can dive onto the live filesystem to see what's what?

Landrash, 2017-02-25:
@kxtcd950 I'm on gitter now if you can go there and I'll get you a copy.
If you got a running setup you can just add it on top of the current image.
Download this on your hassbian install https://github.com/home-assistant/hassbian-scripts/releases/download/v0.2/hassbian-scripts-0.2.deb
install it with `sudo dpkg -i hassbian-scripts-0.2.deb `
And then you should be able to run hassbian-config.
Changes are in the dev branch of this repo :)

kxtcd950, 2017-02-26:
Ok.  I know what the problem is.  Line 30 of hassbian-config (currently ``` suites=get-all-suite-installers```) needs to read ```suites=$(get-all-suite-installers)``` - this slipped through my testing, I'm afraid.

I'm also struggling to pull the dev branch into my fork, so until I've managed to wrangle github and issue a pull request, this'll have to do, sorry :(

kxtcd950, 2017-02-26:
Github mastery improved; dev branch synced into my fork and change made, pull request created.

Landrash, 2017-02-26:
Implemented in dev branch. 
Thanks @kxtcd950 for all your great work :)

closed 2017-02-26 by Landrash