home-assistant/hassbian-scripts #17: Install Python package hassbian

balloob, 2017-02-16:
To detect that we're running hassbian, please install `hassbian` into the image: `pip3 install hassbian`

balloob, 2017-02-16:
Source is here: https://github.com/home-assistant/python-hassbian

Landrash, 2017-02-16:
Could you make a change to the hassbian package to require home-assistant package?
That way we still only install one package and that package installs whatever python packages we want on the pi's?

balloob, 2017-02-16:
I don't like that - the package can run without Home Assistant and by depending on Home Assistant we might block any other tools that people want to write for Hassbian management.

What is wrong with running `pip3 install hassbian-scripts home-assistant` ?

Landrash, 2017-02-17:
Your right. I'll implement your solution :smiley:

Landrash, 2017-02-24:
Implemented in dev branch.

closed 2017-02-24 by Landrash