home-assistant/hassbian-scripts #24: Hassbian not working out of the box

tethyso, 2017-03-15:
Hi there. Just tried a clean install of HASSBIAN 1.1 (2017-02-03 build).

Following given instructions, setup script goes somehow bad: no interface is reachable after 30+ mins of waiting.
Other users with similar issues are suggesting a potential fix
https://community.home-assistant.io/t/hassbian-not-working-out-of-the-box/11479/3

Another problem is that with every sudo commands, I (we) get a 
sudo: unable to resolve host hassbian
which happens when hostname is not in /etc/hosts (and it's not).

Would be nice if this can get fixed, as hassbian is - in my opinion - the more convenient way to start out with home assistant.

Thanks

jonezy199, 2017-03-15:
I had the same problems when I was starting out - now that I've done it 100 times I may be able to help.

First can you walk us through what you did starting from flashing the SD card to running the scripts? Are they the correct scripts from https://github.com/home-assistant/hassbian-scripts ?

And to fix the sudo: unable to blah blah....

open up ssh (putty) and log into your pi.
Type command 'sudo raspi-config' then use the arrow keys to arrow down to "advanced options".  Press Enter. Arrow down to "Host Name".  Press Enter.  Press enter again to past the splash screen for this new window.  Backspace where it says "hassbian" and instead type "raspberrypi". Arrow down to okay. Press enter.  Arrow over to finish.  Press enter.  Press enter to reboot.

Landrash, 2017-03-16:
@tethyso @jonezy199 https://github.com/home-assistant/pi-gen/releases/tag/v1.2 Next release in pre-release. The sudo thing is just an annoyance and something I missed.

@jonezy199 Your fix actually breaks some other things. Best and easiest fix is to change the resolv of the hostname in the hosts file. I'll write up a quick guide if @tethyso needs it.

tethyso, 2017-03-16:
@jonezy199 : no script run, I've just burnt file to sd and booted pi. I can ssh into it, but either the install script does not run on first boot, or home assistant is not started. 

@Landrash : it would be awesome to put guide here https://home-assistant.io/docs/hassbian/installation/. So that everybody can fix the issue. If it's going to bi fixed with 1.2, I guess I'll just wait for it.

Don't get me wrong, there are ways to make hassbian work. But since it's supposed to be a "burn to sd & forget thing", I guess it should just work.

Btw, I'll try to fix hostname issue and run the script again. For now, thanks for your time and help!

Landrash, 2017-03-16:
@tethyso IT should just work. The 1.2 image is ready and you can just use it as is right now. There's been some minor changes to the package and I'm planing on posting it tonight with a blog post on Sathurday.

jonezy199, 2017-03-16:
@Landrash my bad - that seemed to work for my sudo issues when I first booted 1.1

Are the scripts the same in 1.2?

Landrash, 2017-03-17:
@jonezy199 Same and some more. They are now accessed trough a small tool called hassbian-config.
Theres support from HASSbians side to have the scripts launched from within HA but we're waiting for some tooling in HA to be done before it's enabled by default.

```
#Example commands from 
$ hassbian-config show
$ hassbian-config install libcec
 ```

Both commands should be able to be run with either the `pi` or `homeassistant` accounts.

closed 2017-03-17 by Landrash

tethyso, 2017-03-21:
Hi @Landrash !
I've installed image_2017-03-17-HASSbian.zip and confirm Home Assistant is correctly starting.
However a call to homeassistant/restart gives this

`17-03-21 16:05:35 ERROR (MainThread) [homeassistant.loader] Error loading homeassistant.components.http. Make sure all dependencies are installed
Traceback (most recent call last):
  File "/srv/homeassista17-03-21 16:05:35 ERROR (MainThread) [homeassistant.components] INFO:homeassistant.util.package:Attempting install of colorlog>2.1,<3
Testing configuration at /home/homeassistant/.homeassistant
17-03-21 16:05:35 INFO (MainThread) [homeassistant.setup] Setting up introduction
17-03-21 16:05:35 INFO (MainThread) [homeassistant.setup] Setting up conversation
17-03-21 16:05:35 INFO (MainThread) [homeassistant.setup] Setting up updater
17-03-21 16:05:35 INFO (MainThread) [homeassistant.setup] Setting up discovery
17-03-21 16:05:35 ERROR (MainThread) [homeassistant.loader] Error loading homeassistant.components.http. Make sure all dependencies are installed
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/loader.py", line 141, in get_component
    module = importlib.import_module(path)
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/components/http/__init__.py", line 34, in <module>
    from .static import FILE_SENDER, CACHING_FILE_SENDER, staticresource_middleware
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/components/http/static.py", line 6, in <module>
    from aiohttp.file_sender import FileSender
ImportError: No module named 'aiohttp.file_sender'
17-03-21 16:05:35 ERROR (MainThread) [homeassistant.loader] Unable to find component http
17-03-21 16:05:35 ERROR (MainThread) [homeassistant.setup] Setup failed for http: Component not found.
17-03-21 16:05:35 ERROR (MainThread) [homeassistant.loader] Error loading homeassistant.components.tts. Make sure all dependencies are installed
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/loader.py", line 141, in get_component
    module = importlib.import_module(path)
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/components/tts/__init__.py", line 24, in <module>
    from homeassistant.components.http import HomeAssistantView
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/components/http/__init__.py", line 34, in <module>
    from .static import FILE_SENDER, CACHING_FILE_SENDER, staticresource_middleware
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/components/http/static.py", line 6, in <module>
    from aiohttp.file_sender import FileSender
ImportError: No module named 'aiohttp.file_sender'
17-03-21 16:05:35 ERROR (MainThread) [homeassistant.loader] Unable to find component tts
17-03-21 16:05:35 ERROR (MainThread) [homeassistant.setup] Setup failed for tts: Component not found.
17-03-21 16:05:35 INFO (MainThread) [homeassistant.setup] Setting up sun
17-03-21 16:05:35 INFO (MainThread) [homeassistant.setup] Setting up sensor
17-03-21 16:05:35 INFO (MainThread) [homeassistant.setup] Setting up group
17-03-21 16:05:35 ERROR (MainThread) [homeassistant.loader] Error loading homeassistant.components.history. Make sure all dependencies are installed
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/loader.py", line 141, in get_component
    module = importlib.import_module(path)
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
[... 286 lines cut ...]
[... cut ...]
    from .static import FILE_SENDER, CACHING_FILE_SENDER, staticresource_middleware
  File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/components/http/static.py", line 6, in <module>
    from aiohttp.file_sender import FileSender
ImportError: No module named 'aiohttp.file_sender'
17-03-21 16:19:09 ERROR (MainThread) [homeassistant.loader] Unable to find component media_player
17-03-21 16:19:09 ERROR (MainThread) [homeassistant.setup] Setup failed for media_player: Component not found.
`

Please let me know if I can help in any way ;)

Thanks!

jonezy199, 2017-03-21:
@tethyso  go over here https://github.com/home-assistant/home-assistant/issues/6728#issuecomment-287974473  - solution is in discussion.  Looks like it happened to a lot of people last night.

Landrash, 2017-03-22:
@tethyso & @jonezy199 
Fredrik Lindqvist
There's currently an problemwith an updated python package. A hotfix is coming out to HA but there's nothing I can change on the HASSbian side.