home-assistant/hassbian-scripts #33: HASSbian Wishlist

Landrash, 2017-05-08:
Wish list for functionality that we want to add to Hassbian. Pull requests welcome.

### Install scripts
[ ] HomeBridge

### Other functionality
[ ] Hotspot if no network functionality is detected. See #8 for details.

arsaboo, 2017-05-08:
Let'e Encrypt registration and renewal please :pray:

arsaboo, 2017-05-09:
Here are couple more that would be nice to have:
1. MySQLdb
2. Github backup/restore

Landrash, 2017-05-09:
First one I'm unsure about what it is?

arsaboo, 2017-05-09:
Instead of using SQLAlchemy, use MySQLdb for the recorder - https://home-assistant.io/components/recorder/

Landrash, 2017-05-10:
Ahh. The lack of space between MYSQL & db confused me ;)

penright, 2017-07-20:
Thank @Landrash for adding the AppDaemon to the list.
One other Hassbain install that I had to fight through was Open Zwave barrier fixed.
This link starts where I was adding a Linear garage door opener.
[https://community.home-assistant.io/t/linear-ngdz00-4-garage-door/2136/220?u=penright](url)

There was a lot of patches, that most of them I hope will be in the next Hass release.
But there was some patches to the OZW, not sure I fully understand it right now. I kind of did the monkey thing and got it working. The most relevant part to this discussion is when I asked about future patches and this was the question and the answer given.
[image]
[image]

Here is the direct link [https://community.home-assistant.io/t/linear-ngdz00-4-garage-door/2136/235?u=penright](url)

Landrash, 2017-07-20:
@penright Problem with  patch like this is that it won't be persistent between updates. If anything it's something that would need to be fixed in HA itself :slightly_smiling_face:

Landrash, 2017-09-25:
Appdaemon addressed in #52

riemers, 2017-10-29:
I think its good to have a cheat sheet once you login. That is like your motd. With things like stop ha, start ha. All those shortcuts that make it easier.. perhaps add shortcuts "hastop" "hastart" "haupgrade"i do a lot of restarting (i always have to disable z-wave for instance to test new stuff) because it takes 10+ minutes to finally be ready.

Landrash, 2017-10-31:
@riemers 
 - MOTD is on my list but  had forgotten to add it to this one. Adding it
 - Aliases for commonly used commands are on the list already.
 - Having the default location be the `.homeassistant` directory is something I'm not comfortable changing since it makes a change to something that's standard on most *nix installations.

ludeeus, 2018-01-08:
PR #79 for homebridge is ready :)

ardeus-ua, 2018-02-07:
Kiosk mode for connected touchscreen to automatically open Home Assistant. Maybe adaptation of https://github.com/guysoft/FullPageOS

Landrash, 2018-02-08:
@ardeus-ua Out of scope for Hassbian since we don't ship any kind of gui.
Everything is generally done over ssh.

ardeus-ua, 2018-02-08:
I mean the scripted installation of kiosk mode for rpi to open Home Assistant gui on device startup

Landrash, 2018-02-12:
Reopening since this issue should not be closed.

Antoni-K, 2018-02-25:
I'm working on a Webmin installation script for Hassbian right now - could it be added to the wishlist, or should I submit a PR first?

riemers, 2018-02-25:
A use case for webmin would be nice too, i know you can do a lot with it, but it all depends on your needs. Are there any good examples on what you want  to achieve with it? (not saying its a bad idea, just wondering)

Antoni-K, 2018-02-25:
@riemers Webmin makes it really easy to do lots of things you would usually use the CLI for via a GUI, so it could be really useful for beginners. I’ve used it myself in the past, and can confirm that it works fine. AFAIK, you can use it to configure programs like samba or systemd if you’re running services installed from sources other than this (hassbian config).

Landrash, 2018-02-25:
@Antoni-K I don't think it needs to be in the tool.
It's fairly straight forward to install and the instructions are on the webmin download page.
```
wget http://prdownloads.sourceforge.net/webadmin/webmin_1.760_all.deb
apt install webmin_1.760_all.deb
```

The things we added to Hassbian config has been tweaks and tools that are generally used with Home Assistant. There's a point to where we need to keep a balance between functionality and what goals the product is trying to accomplish or well end up creating a new packaging system.

Antoni-K, 2018-02-26:
@Landrash Oh, sorry, didn't see your comment. I submitted a pull request already but feel free to close it. Also, I might get started on an rsync-based backup & restore utility, if nobody else is already working on one.

riemers, 2018-02-26:
Rsync can be cronned, same applies to webmin perhaps. If you do make something with rsync, a better place would be to make a description for it like say https://home-assistant.io/docs/ecosystem/backup/backup_usb/ , that way everybody can enjoy your work regardless of the method they used to install hass.

Antoni-K, 2018-02-26:
Got it. I was thinking of something similar to your USB backup script, but with rsync so it is slightly faster.

riemers, 2018-02-26:
If you have a proper setup, it will only sync a few text files no? I would see the use case rather that you can sync it to another location. Something the USB cannot do :) perhaps link it with a ionotify tool for instant backup, sky is the limit.

Antoni-K, 2018-02-26:
Good idea, maybe also cloud integration (other than dropbox). I'll look into it.

riemers, 2018-02-26:
If you get bored, why not write an installer that asks you how you want your backup and with what options 🥇

Antoni-K, 2018-02-26:
Will do 👍

ferazambuja, 2018-03-26:
Raspberry Pi Desktop :)

Landrash, 2018-03-26:
@ferazambuja No. Out of scope for this project.

ferazambuja, 2018-05-09:
I’m Mac user and I enjoy using http://netatalk.sourceforge.net instead of samba. 
It’s an AppleShare file server. 
Is just more seamless and your hassbian home/pi shows automatically as another Mac. However the home assistant config folder needs to be added manually. It also handles reboots way better avoiding your Atom editor to screen errors if you have config files open. 

It doesn’t exclude the need of samba. But it’s an idea to try to make Hassbian even more accessible.

swetoast, 2018-07-04:
motd is simple we can use neofetch for that looks like this

https://github.com/dylanaraps/neofetch

[image]

riemers, 2018-07-04:
I think the next logical step would be to have the "add-ons" as packages. Most is already like that but then have an option to do it via the GUI, like hassio does it now too. But contain it to one repo only to not get 100's of repository's.. i know i could help out with scripting part but it is the gui part that i have no skills for. Find one of those and i am more then happy to help out on that part too. Imagine a restore which includes the packages you had and it downloads those for you. (might be wanting the world here, but just giving out my thoughts)

Landrash, 2018-07-04:
@ferazambuja 
No way to test since I use a *nix only environment.  Contributions welcome.

@swetoast Neofetch would be great. Minimal custom code from us and that's generally preferred. 

@schmittx I'll add it to the list but nothing I've worked on before.

@riemers Logging into my Hassbian pi, daily generally.
Making packages will require some more changes to the base tools we have written but it can be considered.

riemers, 2018-07-04:
@Landrash that you log in to your PI daily is logical :-) I don't do that so often, use also visual studio code with the yaml plugin and samba mount.

schmittx, 2018-07-20:
How about a Python upgrade script to archive/copy the current venv, install Python, create new venv, etc.?

ardeus-ua, 2018-08-09:
Is this possible to add ffmpeg with hardware decoding to the base image?

Landrash, 2018-08-10:
@ardeus-ua Is it available as a package then we could add it as a package.

Landrash, 2018-11-05:
@nixme Give it a go :) What part of it do you want on the pi?

ardeus-ua, 2018-11-06:
> @ardeus-ua Is it available as a package then we could add it as a package.

https://www.johnvansickle.com/ffmpeg/ here are fresh precompiled binaries
`built with gcc 6.3.0 (Debian 6.3.0-18+deb9u1) 20170516
  configuration: --enable-gpl --enable-version3 --enable-static --disable-debug --disable-ffplay --disable-indev=sndio --disable-outdev=sndio --cc=gcc-6 --enable-fontconfig --enable-frei0r --enable-gnutls --enable-gray --enable-libfribidi --enable-libass --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-librubberband --enable-libsoxr --enable-libspeex --enable-libvorbis --enable-libopus --enable-libtheora --enable-libvidstab --enable-libvo-amrwbenc --enable-libvpx --enable-libwebp --enable-libx264 --enable-libxml2 --enable-libxvid --enable-libzimg`

pepeEL, 2019-02-04:
hi
good will be if you add Backup and restore...

dasb00ter, 2019-03-20:
I would like to make a request for a good reverse proxy. I am finding the install daunting but it is security related so I feel it would be valuable. If a script could get everything installed in the right places I could probably cobble together a how to wiki for it once I get it running myself. I guess Caddy or NGINX would be good candidates with the former I see being utilized in the HASS.IO addons and therefor perhaps more user friendly

jurgenweber, 2019-07-24:
can i suggest a flag where you can tell the home assistant suite not to restart on upgrade?

ludeeus, 2019-07-24:
If you want that use pip manually.

Landrash, 2019-10-26:
https://www.home-assistant.io/blog/2019/10/26/rip-hassbian/

PR #107 (2018-02-11): Aliases for common functions.
### Description:
Adds aliases for common functions.

**Related issue (if applicable):** Fixes #33  

### Checklist:
  - [x] The code change is tested and works locally.
  - [ ] Script has validation check of the job.
  
#### If pertinent:
  - [ ] Created/Updated documentation at `/docs`
closed 2019-10-26 by Landrash