home-assistant/hassbian-scripts #36: New HASSBIAN install, hassbian-config install openzwave

jkvalence, 2017-06-15:
Hello,

Just started a new installation with HASSBIAN 1.23.  Ran into an issue that I wanted to make you aware of.  Below are the steps I took during the install and the error from my hass.log.

1.  Finish HASSBIAN installation
2.  Set raspi-config options
3.  Rebooted
4.  Updated (update/upgrade)
5.  Rebooted
6.  Sudo hassbain-config install samba
7.  Rebooted
8.  Sudo hassbian-config install openzwave
9.  Rebooted
10.  Tried to access HA from browser, no luck, ran hass from putty, stalled trying to install OZWCP, rebooted
11.  HASS launched after reboot with zwave config error
12.  Error log is as follows.

> 2017-06-15 01:29:55 ERROR (MainThread) [homeassistant.setup] Error during setup of component zwave
> Traceback (most recent call last):
>   File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/setup.py", line 190, in _async_setup_component
>     component.setup, hass, processed_config)
>   File "/usr/lib/python3.4/asyncio/futures.py", line 388, in __iter__
>     yield self  # This tells Task to wait for completion.
>   File "/usr/lib/python3.4/asyncio/tasks.py", line 286, in _wakeup
>     value = future.result()
>   File "/usr/lib/python3.4/asyncio/futures.py", line 277, in result
>     raise self._exception
>   File "/usr/lib/python3.4/concurrent/futures/thread.py", line 54, in run
>     result = self.fn(*self.args, **self.kwargs)
>   File "/srv/homeassistant/lib/python3.4/site-packages/homeassistant/components/zwave/__init__.py", line 258, in setup
>     config_path=config[DOMAIN].get(CONF_CONFIG_PATH))
>   File "/srv/homeassistant/lib/python3.4/site-packages/pyozwweb-0.4.0.35-py3.4.egg/openzwave/option.py", line 82, in __init__
>     libopenzwave.PyOptions.__init__(self, config_path=config_path, user_path=user_path, cmd_line=cmd_line)
>   File "src-lib/libopenzwave/libopenzwave.pyx", line 669, in libopenzwave.PyOptions.__init__ (openzwave-embed/open-zwave-master/python-openzwave/src-lib/libopenzwave/libopenzwave.cpp:10816)
> libopenzwave.LibZWaveException: "LibOpenZwave Generic Exception : Can't autoconfigure path to config"
> 

Frederick, normally I would contact you directly but seeing as how you are active with both HASSBIAN and these config files, I wanted to let you know asap.  There are no other changes I made to the Pi or Hassbian install so there shouldn't have been any steps I missed.

jkvalence

Landrash, 2017-06-22:
The openzwave install script isn't needed anymore. It's installed automatically. by default in Home Assistant.

The install doesn't stall but takes quite some time. (1 hour+) You rebooting in the middle of it kinda broke it and now I'm unsure how you would fix it actually.

jkvalence, 2017-06-23:
Frederick,

I appreciate your response.  I gave up on HASSbian and went with the AIO installer and it went off without errors.  Again, I appreciate your work with HA.  In the future, if I find anything I will message you directly.  Feel free to delete this issue from the post as it is no longer relevant to either of us.

Thank you again Frederick!

closed 2017-06-25 by Landrash