home-assistant/hassbian-scripts #38: error installing libcec using hassbian-config

willie, 2017-07-19:
Starting with 2017-05-09-Hassbian.img on a Raspberry Pi 3, I logged in, did the following:

```
sudo apt-get update 
sudo apt-get upgrade 
reboot
```
Logged back in and proceeded to try and install libcec using hassbian-config. Errors ensued.

Any suggestions to resolve this problem?

```
pi@hassbian:/home/homeassistant/.homeassistant $ hassbian-config install libcec
/usr/local/bin/hassbian-config: line 47: /srv/homeassistant/hassbian/suite-states.lock: Permission denied
flock: 300: Bad file descriptor
pi@hassbian:/home/homeassistant/.homeassistant $ sudo !!
sudo hassbian-config install libcec
libcec install script for Hassbian
Copyright(c) 2017 Fredrik Lindqvist <https://github.im/Landrash>
Running apt-get preparation
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages                            
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages                           
Hit http://archive.raspberrypi.org jessie/ui armhf Packages               
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB                                 
Ign http://archive.raspberrypi.org jessie/main Translation-en                                       
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB           
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en              
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB                    
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en                 
Ign http://archive.raspberrypi.org jessie/ui Translation-en                       
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  cmake-data libarchive13 libjsoncpp0 liblzo2-2 libpthread-stubs0-dev libpython-dev libpython2.7-dev libx11-dev libx11-doc libxau-dev libxcb1-dev libxdmcp-dev
  libxext-dev libxrandr2 libxrender-dev libxrender1 python2.7-dev swig2.0 x11proto-core-dev x11proto-input-dev x11proto-kb-dev x11proto-randr-dev
  x11proto-render-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
Suggested packages:
  codeblocks eclipse ninja-build lrzip libxcb-doc libxext-doc swig-doc swig-examples swig2.0-examples swig2.0-doc
The following NEW packages will be installed:
  cmake cmake-data libarchive13 libjsoncpp0 liblzo2-2 libpthread-stubs0-dev libpython-dev libpython2.7-dev libudev-dev libx11-dev libx11-doc libxau-dev
  libxcb1-dev libxdmcp-dev libxext-dev libxrandr-dev libxrandr2 libxrender-dev libxrender1 python-dev python2.7-dev swig swig2.0 x11proto-core-dev
  x11proto-input-dev x11proto-kb-dev x11proto-randr-dev x11proto-render-dev x11proto-xext-dev xorg-sgml-doctools xtrans-dev
0 upgraded, 31 newly installed, 0 to remove and 0 not upgraded.
Need to get 28.2 MB of archives.
After this operation, 67.9 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian/ jessie/main cmake-data all 3.6.2-2~bpo8+1 [1,168 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main liblzo2-2 armhf 2.08-1.2 [47.6 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main libarchive13 armhf 3.1.2-11+deb8u3 [231 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian/ jessie/main libjsoncpp0 armhf 0.6.0~rc2-3.1 [61.1 kB]
Get:5 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpython2.7-dev armhf 2.7.9-2+deb8u1 [17.9 MB]
Get:6 http://archive.raspberrypi.org/debian/ jessie/main cmake armhf 3.6.2-2~bpo8+1 [2,379 kB]     
Get:7 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-dev armhf 2.7.9-1 [1,188 B]
Get:8 http://archive.raspberrypi.org/debian/ jessie/main x11proto-core-dev all 7.0.31-1 [728 kB]                                                                
Get:9 http://archive.raspberrypi.org/debian/ jessie/main xtrans-dev all 1.3.5-1 [100 kB]                                                                        
Get:10 http://archive.raspberrypi.org/debian/ jessie/main x11proto-randr-dev all 1.5.0-1 [49.7 kB]                                                              
Get:11 http://mirrordirector.raspbian.org/raspbian/ jessie/main libxrender1 armhf 1:0.9.8-1 [30.4 kB]                                                           
Get:12 http://mirrordirector.raspbian.org/raspbian/ jessie/main libxrandr2 armhf 2:1.4.2-1 [31.7 kB]                                                            
Get:13 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpthread-stubs0-dev armhf 0.3-4 [4,042 B]                                                     
[... 277 lines cut ...]
-- Found SWIG: /usr/bin/swig2.0 (found version "2.0.12") 
CMake Error at src/libcec/cmake/CheckPlatformSupport.cmake:193 (if):
  if given arguments:

    "3" "GREATER" "2" "AND" "3" "GREATER_EQUAL" "7"

  Unknown arguments specified
Call Stack (most recent call first):
  src/libcec/CMakeLists.txt:133 (include)

-- Configuring incomplete, errors occurred!
See also "/srv/homeassistant/src/libcec/build/CMakeFiles/CMakeOutput.log".
See also "/srv/homeassistant/src/libcec/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
Installing Pulse-Eight libcec
make: *** No rule to make target 'install'.  Stop.
Linking libcec to venv site packages

Installation done.

If you have issues with this script, please contact @Landrash on gitter.im

To continue have a look at https://home-assistant.io/components/hdmi_cec/
It's recomended that you restart your Pi before continuing with testing libcec.
```

gilhaz, 2017-07-20:
Same here, I get the same error..

Landrash, 2017-07-20:
Problem is upstream with one of the dependencies(swig). I'll have to try and find a work around when I got the time.

gilhaz, 2017-07-21:
I spent all night on this thing with no luck.. :(
My home is dumb now..

gilhaz, 2017-07-25:
# **Finally!** 
If anyone is still need it..
**I Mannaged to install libCEC** by removing lines `191-199` in the file:
`~/libcec/src/libcec/cmake/CheckPlatformSupport.cmake`

## This is my new installation process:
```
sudo apt-get update
sudo apt-get install cmake libudev-dev libxrandr-dev python-dev swig
cd
git clone https://github.com/Pulse-Eight/platform.git
mkdir platform/build
cd platform/build
cmake ..
make
sudo make install
cd
git clone https://github.com/Pulse-Eight/libcec.git
```
`sed -i '191,199d' ~/libcec/src/libcec/cmake/CheckPlatformSupport.cmake`
* _If you want to manually edit the file see below._
```
mkdir libcec/build
cd libcec/build
cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..
make -j4
sudo make install
sudo ldconfig
```
_**Test it**_
`echo scan | cec-client -s -d 1`

_____________________
_____________________
**_To manually edit `CheckPlatformSupport.cmake`_**
`sudo nano ~/libcec/src/libcec/cmake/CheckPlatformSupport.cmake`
Remove those lines:
```
    SET(PYTHON_LIB_INSTALL_PATH "/cec" CACHE STRING "python lib path")
    if (${CMAKE_MAJOR_VERSION} GREATER 2 AND ${CMAKE_MAJOR_VERSION} GREATER_EQUAL 7)
          SET(PYTHON_LIB_INSTALL_PATH "" CACHE STRING "python lib path" FORCE)
    else()
      if (${CMAKE_MAJOR_VERSION} GREATER_EQUAL 3)
        SET(PYTHON_LIB_INSTALL_PATH "" CACHE STRING "python lib path" FORCE)
      endif()
    endif()
```
_____________________
_____________________
Source: [Pulse-Eight/libcec/issues/362](https://github.com/Pulse-Eight/libcec/issues/362#issuecomment-317258888)
By: @john-ruddy

andy380743909, 2017-08-02:
save me tons of time

dmikhalsky, 2017-08-16:
@gilhaz Thanxxxxx!!!!

Landrash, 2017-09-05:
Changes implemented in dev branch but other problem exist with the libcec for newer versions of python.

closed 2017-09-05 by Landrash

DeMuizen, 2017-09-10:
thank you very much!

stevenazari, 2017-09-21:
still broken for me X(

will try the fix from @gilhaz tonight

Landrash, 2017-09-21:
@stevenazari This is the repository for the hassbian-scripts package that's only is used for hassbian. There's a separate repository for the all-in-on-installer.