home-assistant/hassbian-scripts #39: Hassbian Strech

ludeeus, 2017-08-21:
All issues are experienced on the [1.3 WIP.](https://github.com/home-assistant/pi-gen/releases/tag/1.3)

[  ] **sudo hassbian-config install mosquitto _failed_**: Missing dependicies (libssl & libwebsockets)
[  ] **sudo hassbian-config install hue _failed_**: Worng python version, should be 3.5 not 3.4
[  ] **sudo hassbian-config install tellstick _failed_**: libconfuse0 (>= 2.5) but it is not installable
[  ] **sudo hassbian-config install libcec _failed_**: Network crashed during setup.
[  ] **Unable to install package pyatv==0.3.4**

ludeeus, 2017-08-21:
Added this #40 for hue.

Landrash, 2017-08-22:
Libcec fixed in stretch branch.
Hue fixed in stretch branch and an alternative fix has been submitted by @ludeeus in #40 . Considering implementing that fix since it's a better option.
Openzwave & pyAtv are WIP since some changes are needed to the base system since dependencies has changed.
Mosquitto is almost done but not properly tested.

Tellstick will be removed since packages are not available for Tellstick.

Landrash, 2017-09-25:
Closing after release

closed 2017-09-25 by Landrash

hnnweb, 2017-11-06:
Tellstick works in Stretch

https://community.home-assistant.io/t/home-assistant-running-on-raspberry-pi-2-b-with-tellstick-duo/4346/12

Landrash, 2017-11-07:
@hnnweb Feel free to submit a new install script for it. I'll try to do it if not when I got the time but that might take a while.