home-assistant/hassbian-scripts #54: can't find install tellstick command

fredriklindell, 2017-10-05:
Hi,

Reading this guidehttps://home-assistant.io/docs/installation/hassbian/customization/

Tells me that there should be a command like "hassbian-config install tellstick", is it in development/removed?

Thanks in advance!

Landrash, 2017-10-05:
@fredriklindell Hi Fredrik. 
The install script for Tellstick was removed with Debian Stretch since the package and dependencies haven't been updated by Telldus.

Repackaging and updating that package is out of scope for Hassbian so support was regrettably dropped. I do realize now that we have missed to document that in the release notes and i'll augment that as soon a possible.

Landrash, 2017-10-05:
For some further reading [here's](https://github.com/home-assistant/hassbian-scripts/issues/39) the Issued we tracked the development for Hassbian Stretch

closed 2017-10-05 by Landrash

jonne013, 2017-10-09:
Hi

I can gladly report that I've gotten Tellstick to run on Debian Stretch.

To begin with I found out that there are a beta version of telldus-core available through this [ticket](http://developer.telldus.se/ticket/473#comment:2).
On this other repository i found the source for the beta version [direct link](https://s3.eu-central-1.amazonaws.com/download.telldus.com/pool/main/t/telldus-core/telldus-core_2.1.3-beta1.orig.tar.gz).
From there i built and installed it by following this [guide](https://developer.telldus.com/wiki/TellStickInstallationSource). And in order to make it start as a service using systemctl i found this [forum post](https://forum.telldus.com/viewtopic.php?f=8&t=5625#p31963).

I am not proficient enough to make a script out of this, but I hope I can point someone who is in the right direction.

Kind regards