home-assistant/hassbian-scripts #55: sudo hassbian-config install webterminal

juan11perez, 2017-10-05:
This command returns:
"suite webterminal doesn't exist."

Landrash, 2017-10-05:
@juan11perez Tested on the latest image available. Install functions as intended. 
When did you install Hassbian? Most likely you are on a older version.

Output from Hassbian 1.3 below:
```shell
pi@hassbian:~ $ hassbian-config show
Install scripts: (Usage 'sudo hassbian-config install suite-name')
mosquitto:           Mosquitto Installer for Hassbian
webterminal:         Installs an websevice terminal.
mariadb:             MariaDB install script for Hassbian
tradfri:             Tradfri install script for Hassbian
hue:                 Echo/Home/Mycroft Emulated Hue install script for Hassbian
samba:               Samba install script for Hassbian
appdaemon:           AppDaemon install script for Hassbian
libcec:              libcec install script for Hassbian
duckdns:             Setup for Duck DNS auto renewal.

Upgrade scripts: (Usage 'sudo hassbian-config upgrade suite-name')
hassbian:            Home Assistant upgrade script for Hassbian
hassbian-script-dev: Uppdates scripts from the dev branch.
hassbian-script:     Hassbian-Script upgrade script for Hassbian
home-assistant:      Home Assistant upgrade script for Hassbian

To view more info about a suite type: 'hassbian-config show suite-name'
pi@hassbian:~ $ hassbian-config show webterminal
Installs an websevice terminal.
Installs an websevice terminal to controll your installation.
Original consept by Ludeeus <https://github.com/ludeeus>
pi@hassbian:~ $ sudo hassbian-config install webterminal
Installing packages.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
openssl is already the newest version (1.1.0f-3).
openssl set to manually installed.
The following NEW packages will be installed:
  shellinabox
0 upgraded, 1 newly installed, 0 to remove and 6 not upgraded.
Need to get 116 kB of archives.
After this operation, 496 kB of additional disk space will be used.
Get:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf shellinabox armhf 2.20 [116 kB]
Fetched 116 kB in 0s (325 kB/s)    
apt-listchanges: Can't set locale; make sure $LC_* and $LANG are correct!
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_TIME = "sv_SE.UTF-8",
	LC_MONETARY = "sv_SE.UTF-8",
	LC_ADDRESS = "sv_SE.UTF-8",
	LC_TELEPHONE = "sv_SE.UTF-8",
	LC_NAME = "sv_SE.UTF-8",
	LC_MEASUREMENT = "sv_SE.UTF-8",
	LC_IDENTIFICATION = "sv_SE.UTF-8",
	LC_NUMERIC = "sv_SE.UTF-8",
	LC_PAPER = "sv_SE.UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_ALL to default locale: No such file or directory
Selecting previously unselected package shellinabox.
(Reading database ... 41447 files and directories currently installed.)
Preparing to unpack .../shellinabox_2.20_armhf.deb ...
Unpacking shellinabox (2.20) ...
Setting up shellinabox (2.20) ...
Processing triggers for systemd (232-25+deb9u1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Changing config.
Reloading and starting the service.

Installation done.

You can now access the web terminal here: http://192.168.0.8:4200
You can also add this to your Home-Assistant config in an 'panel_iframe'

If you have issues with this script, please say something in the #Hassbian channel on Discord.
```

closed 2017-10-05 by Landrash

Landrash, 2017-10-05:
My bad. It's in the dev release and not yet released.

Run the following to get it currently
```bash
sudo curl -o /opt/hassbian/suites/install_webterminal.sh https://raw.githubusercontent.com/home-assistant/hassbian-scripts/dev/package/opt/hassbian/suites/install_webterminal.sh && sudo hassbian-config install webterminal
```

juan11perez, 2017-10-05:
Yes Fred.
Many thanks for your help.

Regards
Juan