home-assistant/hassbian-scripts #63: Installation fails without sources repo enabled

Nedlinin, 2017-10-10:
Installation on Hassbian install of ~6 months old.

/etc/apt/sources.list did not have the sources bit uncommented.

    deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
    # Uncomment line below then 'apt-get update' to enable 'apt-get source'
    #deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi

Had to uncomment it to allow installation.

    deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
    # Uncomment line below then 'apt-get update' to enable 'apt-get source'
    deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi

Maybe should add to install guide?

Landrash, 2017-10-12:
Image has moved away from Jessie and is now stretch based.
Hassbian is the image we distribute and `hassbian-scripts` that I assume your installing on requires packages on only available on stretch since Debian has broken the naming of some packages.

What install guide are your referring to?

Nedlinin, 2017-10-12:
@Landrash :  Can't say I can speak toward the Debian side of things.  But, I can say that while on the older image (Jessie based) if I made that edit to allow downloading of sources/dev packages the package would install fine.

There is a make_install.sh script.  I think part of the issue is instead of cloning the repo and running this script on a Linux box on the same local network, I cloned the repo to my Hassbian install directly and ran said script directly on the RPi3.

The RPi3 (with Hassbian) did not have the proper sources available for package dependencies during installation and failed out.  Enabled the dev/sources repo as I mentioned above, doing an update of sources and then attempting the install again fixed the issue.

I unfortunately didn't save the error for which 3 packages it failed dependencies on but one of them was https://packages.debian.org/jessie/libffi-dev.

This is probably less of an issue for people who end up on 0.6 already as it has an upgrade option built in.  

    > What install guide are your referring to?

My comment should have read "maybe should add *an* install guide?"  Apologies for the miscommunication there.  Even just a note saying "hey make sure you have sources enabled before installing" would be beneficial but updating the make_install.sh to handle it for you would be great I think.

Landrash, 2017-10-13:
@Nedlinin Sounds doable.

How about a compromise currently?
We add the information above about using the package with Raspbian Jessie to the end of the [Readme](https://github.com/home-assistant/hassbian-scripts/blob/dev/README.md)?

Want to do it yourself and work towards https://hacktoberfest.digitalocean.com/?

Landrash, 2017-10-29:
Added to dev branch.

closed 2017-10-29 by Landrash