home-assistant/hassbian-scripts #66: hassbian-config state is not working

ludeeus, 2017-10-14:
On this PRi I have installed tradfri and Hue, both are showing as "Uninstalled"

Output:
```
pi@hassbian:~ $ sudo hassbian-config state
{ "suites": { "appdaemon": { "state": "Uninstalled", "description": "AppDaemon install script for Hassbian" } , "duckdns": { "state": "Uninstalled", "description": "Setup for Duck DNS auto renewal." } , "hue": { "state": "Uninstalled", "description": "Echo/Home/Mycroft Emulated Hue install script for Hassbian." } , "libcec": { "state": "Uninstalled", "description": "libcec install script for Hassbian" } , "mariadb": { "state": "Uninstalled", "description": "MariaDB install script for Hassbian" } , "mosquitto": { "state": "Uninstalled", "description": "Mosquitto Installer for Hassbian" } , "mssql": { "state": "Uninstalled", "description": "MS SQL install script for Hassbian" } , "mysql": { "state": "Uninstalled", "description": "MySQL install script for Hassbian" } , "postgresql": { "state": "Uninstalled", "description": "PostgreSQL install script for Hassbian" } , "razberry": { "state": "Uninstalled", "description": "Disables Bluetooth for the use of a RaZberry with Hassbian." } , "samba": { "state": "Uninstalled", "description": "Samba install script for Hassbian" } , "tradfri": { "state": "Uninstalled", "description": "Tradfri install script for Hassbian" } , "webterminal": { "state": "Uninstalled", "description": "Installs an websevice terminal." } } }

```

Landrash, 2017-10-18:
Your suggested workaround in discord is a good option.

Landrash, 2018-01-28:
Closed with #68

closed 2018-01-28 by Landrash