home-assistant/hassbian-scripts #73: hassbian-config is removed

hnnweb, 2017-11-12:
hassbian-config gets removed when running `apt autoremove`

hnnweb, 2017-11-12:
And when trying to reinstall

``
hnn@hassbian:~ $ sudo apt install hassbian-scripts/now
Reading package lists... Done
Building dependency tree
Reading state information... Done
Selected version '0.6' ( [all]) for 'hassbian-scripts'
The following additional packages will be installed:
  bluetooth git git-man libbluetooth-dev libbluetooth3 liberror-perl libffi-dev libpython-dev libpython2.7 libpython2.7-dev libssl-dev libssl-doc
Suggested packages:
  bluez-cups bluez-obexd git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-arch git-cvs git-mediawiki git-svn
E: Can't find a source to download version '0.6' of 'hassbian-scripts:armhf'
``

Landrash, 2017-11-14:
Side effect of not being installed from a repository. Don't have a simple solution currently so I'll leave this issue open.

Thanks for reporting this :cake:

ludeeus, 2017-11-14:
I can't replicate `apt autoremove`
[image]
This is from an fresh install of the newest image.

hnnweb, 2017-11-14:
@ludeeus You need to have system tell you to use autoremove.

Usually this happens when you upgrade or removes packages.
For example, install a new kernel. Kernels are not upgrade as normal packages, they are installed as new ones.

An example of the message:

# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... The following packages were automatically installed and are no longer required:
  bluetooth bluez-cups libboost-filesystem1.55.0
  libboost-program-options1.55.0 libdumbnet1 libgssapi3-heimdal
  libheimntlm0-heimdal libkdc2-heimdal powertop zerofree
Use 'apt-get autoremove' to remove them.
Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

ludeeus, 2017-11-14:
@hnnweb I see :)
Got it to ask now, but I still have no luck replication the removal :(

clin407, 2017-12-10:
I have hassbian-config missing somehow as well.
```
pi@hassbian:/ $ sudo hassbian-config upgrade home-assistant
sudo: hassbian-config: command not found```

whats the best way to put it back where it belongs?

hnnweb, 2017-12-11:
@clin407 Clone this repo.

Adjust the file `make_package.sh` and to:

``#!/bin/bash
dpkg-deb --build package/
sudo dpkg -i package.deb``

Landrash, 2017-12-11:
Download the deb file and install it once again. :)

hnnweb, 2017-12-11:
@Landrash 

From? ;)

Landrash, 2017-12-11:
@hnnweb https://github.com/home-assistant/hassbian-scripts/releases

ludeeus, 2018-01-08:
This should be fixed by implementing the distribution method mentioned in #80

closed 2018-01-28 by Landrash