home-assistant/hassbian-scripts #74: Webterminal Hassbian Script Over HTTPS

carlostico, 2017-11-18:
Webterminal Script for Hassbian is great but it does not work over HTTPS 
The script uses shellinabox and that has support for ssl  ... anyone ??

ludeeus, 2017-11-28:
I now have it working in Firefox, but hadd to add an exeption due to missing cert.
[image]

In chrome I get this:
[image]

If I expose the port, I could probably use my LetsEncrypt Cert, but I do not want ssh to be open from outside my network.

There is an option for self-signed cert, but Chrome don't like those..

@Landrash, do you have some idea on how to takle this?

Landrash, 2017-11-28:
@ludeeus You need to add that exception since it from "another domain" in chrome.

You could use the same Cert on both of them.

[This](https://github.com/shellinabox/shellinabox/pull/411/files) could be of some help to add the Let's Encrypt cert.

Landrash, 2018-01-28:
Closing since this is an enchantment that needs to be included upstream.  https://github.com/shellinabox/shellinabox/pull/411 is suggested as a fix but has yet to be merged.

closed 2018-01-28 by Landrash

juan11perez, 2018-02-19:
@ludeeus 
After days of trial and error I found a way to get Shellinabox on https using the letsencrypt certificate:
Enabled ssl by confirming sudo nano /etc/default/shellinabox with:
SHELLINABOX_ARGS="--no-beep" 
I then made a copy of my fullchain.pem in /etc/letsencrypt/live/mydomain.duckdns.org. 
to that copy which I named certificate-mydomain.duckdns.org.pem I added at the top the content of my privkey.pem.
I then created a symlink as such:
sudo ln -sf /etc/letsencrypt/live/mydomain.duckdns.org/certificate-juan11perez.duckdns.org.
pem /var/lib/shellinabox/certificate-mydomain.duckdns.org.pem
restarted shellinabox - sudo invoke-rc.d shellinabox restart
and opened in chrome. no errors or warnings. see below

[image]

PR #130 (2018-03-13): Webterminal: Added SSL support with existing LE certificates.
## Description:
This add the option to use existing Let's Encrypt certificates.

Tested with:
- [X] Generated with Dehydrated. (DuckDNS script) tested by @ludeeus
- [X] Generated with certbot. _Simulated testing bt @ludeeus, se comment, this should work._

**Related issue (if applicable):** Fixes #74 

## Checklist:
  - [X] The code change is tested and works locally.
  - [X] Script has validation check of the job.
  
### If pertinent:
  - [X] Created/Updated documentation at `/docs`
  
[image]