home-assistant/hassbian-scripts #75: Mosquitto - Illegal instruction

Antexa, 2017-11-27:
Hi, 

I wanted to know if you got the same behavior on your installation of mosquitto. I got definitely an issue with this... 

```
pi@hassbian:~ $ mosquitto
Illegal instruction
pi@hassbian:~ $ sudo mosquitto
Illegal instruction
pi@hassbian:~ $ sudo -i
root@hassbian:~# mosquitto
Illegal instruction

root@hassbian:~# uname -a
Linux hassbian 4.9.41+ #1023 Tue Aug 8 15:47:12 BST 2017 armv6l GNU/Linux

Sorry if you have already covered it...

ludeeus, 2017-11-27:
The workarond for stretch seems to fail on RPi Zero boards:
https://github.com/eclipse/mosquitto/issues/529

ludeeus, 2018-01-04:
@Antexa Do you still have this issue?
If you do, can you try this:
```shell
sudo apt update
sudo apt upgrade
```
And after that `sudo hassbian-config install mosquitto`.

Landrash, 2018-01-28:
Should be fixed with new release. Closing since this is issue is stale and should be reported upstream to the Mosquitto project if it persists.