home-assistant/hassbian-scripts #76: Installation script not working.

lassivv, 2017-12-17:
Installing script not work with new Hassbian installation.

`pi@hassbian:~ $ sudo hassbian-config install mosquitto
Mosquitto Installer for Hassbian
Copyright(c) 2016 Dale Higgs <https://github.com/dale3h>
Modified by Landrash for use with Hassbian.
Adding mosquitto user
The system user `mosquitto' already exists. Exiting.
Creating pid file
Creating data directory
Installing repository key
--2017-12-17 15:50:25--  http://repo.mosquitto.org/debian/mosquitto-repo.gpg.key
Resolving repo.mosquitto.org (repo.mosquitto.org)... 85.119.83.194, 2001:ba8:1f1:f271::2
Connecting to repo.mosquitto.org (repo.mosquitto.org)|85.119.83.194|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3137 (3.1K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                      100%[===========================================================================>]   3.06K  --.-KB/s    in 0s

2017-12-17 15:50:30 (76.8 MB/s) - written to stdout [3137/3137]

OK
Adding repository
--2017-12-17 15:50:30--  http://repo.mosquitto.org/debian/mosquitto-stretch.list
Resolving repo.mosquitto.org (repo.mosquitto.org)... 85.119.83.194, 2001:ba8:1f1:f271::2
Connecting to repo.mosquitto.org (repo.mosquitto.org)|85.119.83.194|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 50 [application/octet-stream]
Saving to: ‘mosquitto-stretch.list.3’

mosquitto-stretch.list.3               100%[===========================================================================>]      50  --.-KB/s    in 0s

2017-12-17 15:50:35 (1.59 MB/s) - ‘mosquitto-stretch.list.3’ saved [50/50]

Installing mosquitto
Hit:1 http://repo.mosquitto.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:3 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
N: Ignoring file 'mosquitto-stretch.list.1' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
N: Ignoring file 'mosquitto-stretch.list.3' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
N: Ignoring file 'mosquitto-stretch.list.2' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mosquitto : Depends: libssl1.0.0 (>= 1.0.1) but it is not installable
             Depends: libwebsockets3 (>= 1.2) but it is not installable
 mosquitto-clients : Depends: libmosquitto1 (= 1.4.12-0mosquitto1) but it is not going to be installed
N: Ignoring file 'mosquitto-stretch.list.1' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
N: Ignoring file 'mosquitto-stretch.list.3' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
N: Ignoring file 'mosquitto-stretch.list.2' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
E: Unable to correct problems, you have held broken packages.
First try failed, adding dependencies and trying again.
This is an workaround and will be omited once it's fixed upstream.
Downloading dependencies
--2017-12-17 15:50:59--  http://ftp.se.debian.org/debian/pool/main/o/openssl/libssl1.0.0_1.0.1t-1+deb8u6_armhf.deb
Resolving ftp.se.debian.org (ftp.se.debian.org)... 194.71.11.173, 194.71.11.165, 2001:6b0:19::173, ...
Connecting to ftp.se.debian.org (ftp.se.debian.org)|194.71.11.173|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-12-17 15:51:04 ERROR 404: Not Found.

--2017-12-17 15:51:04--  http://ftp.se.debian.org/debian/pool/main/libw/libwebsockets/libwebsockets3_1.2.2-1_armhf.deb
Resolving ftp.se.debian.org (ftp.se.debian.org)... 194.71.11.173, 194.71.11.165, 2001:6b0:19::173, ...
Connecting to ftp.se.debian.org (ftp.se.debian.org)|194.71.11.173|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 38396 (37K) [application/x-debian-package]
Saving to: ‘libwebsockets3_1.2.2-1_armhf.deb’

libwebsockets3_1.2.2-1_armhf.deb       100%[===========================================================================>]  37.50K  --.-KB/s    in 0.02s

2017-12-17 15:51:09 (1.78 MB/s) - ‘libwebsockets3_1.2.2-1_armhf.deb’ saved [38396/38396]

Installing dependencies
dpkg: error: cannot access archive 'libssl1.0.0_1.0.1t-1+deb8u6_armhf.deb': No such file or directory
Selecting previously unselected package libwebsockets3:armhf.
(Reading database ... 41062 files and directories currently installed.)
Preparing to unpack libwebsockets3_1.2.2-1_armhf.deb ...
Unpacking libwebsockets3:armhf (1.2.2-1) ...
dpkg: dependency problems prevent configuration of libwebsockets3:armhf:
 libwebsockets3:armhf depends on libssl1.0.0 (>= 1.0.0); however:
  Package libssl1.0.0 is not installed.

dpkg: error processing package libwebsockets3:armhf (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libwebsockets3:armhf
Cleanup dependencies
rm: cannot remove 'libssl1.0.0_1.0.1t-1+deb8u6_armhf.deb': No such file or directory
Retrying installation of mosquitto
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libwebsockets3 : Depends: libssl1.0.0 (>= 1.0.0) but it is not installable
 mosquitto : Depends: libssl1.0.0 (>= 1.0.1) but it is not installable
 mosquitto-clients : Depends: libmosquitto1 (= 1.4.12-0mosquitto1) but it is not going to be installed
N: Ignoring file 'mosquitto-stretch.list.1' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
N: Ignoring file 'mosquitto-stretch.list.3' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
N: Ignoring file 'mosquitto-stretch.list.2' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
Writing default configuration
/opt/hassbian/suites/install_mosquitto.sh: line 72: cd: /etc/mosquitto: No such file or directory
mv: cannot stat 'mosquitto.conf': No such file or directory
cp: cannot create regular file '/etc/mosquitto/mosquitto.conf': No such file or directory
chown: cannot access 'mosquitto.conf': No such file or directory
Initializing password file

Please take a moment to setup your first MQTT user
If no choice is made a default account will be created`

ludeeus, 2017-12-17:
I managed to replicate this issue on an new install, one of the packages introduced in the workaround #47 is no longer available, and the script fails..

```
Downloading dependencies
--2017-12-17 15:50:59-- http://ftp.se.debian.org/debian/pool/main/o/openssl/libssl1.0.0_1.0.1t-1+deb8u6_armhf.deb
Resolving ftp.se.debian.org (ftp.se.debian.org)... 194.71.11.173, 194.71.11.165, 2001:6b0:19::173, ...
Connecting to ftp.se.debian.org (ftp.se.debian.org)|194.71.11.173|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-12-17 15:51:04 ERROR 404: Not Found.
```

ludeeus, 2017-12-17:
Workaround:
```
sudo curl -o /opt/hassbian/suites/install_mosquitto.sh https://gist.githubusercontent.com/ludeeus/7325db650da394401741c1db64491beb/raw/49aa38d66fe9ce2d2002af950046b36e4b2bfeb4/install_mosquitto.sh && sudo hassbian-config install mosquitto
```

Landrash, 2017-12-18:
Solved with #77 .
Won't be available until next release or if the development package is used.

closed 2017-12-18 by Landrash

dpritt, 2017-12-31:
I too have hit this problem, tried your workaround and I am getting the following errors

Err:1 http://repo.mosquitto.org/debian stretch/main armhf libmosquitto1 armhf 1.                                                          4.12-0mosquitto1
  404  Not Found [IP: 85.119.83.194 80]
Err:2 http://repo.mosquitto.org/debian stretch/main armhf mosquitto armhf 1.4.12                                                          -0mosquitto1
  404  Not Found [IP: 85.119.83.194 80]

The script is referencing the incorrect version, this has now moved on to 1.4.14.

ludeeus, 2018-01-02:
@dpritt The workaround is introduced here is for an other issue, and wont help with the current issue.
If you need to run mosquitto now, you can do this by running this:

```
sudo curl -o /opt/hassbian/suites/install_mosquitto.sh https://raw.githubusercontent.com/ludeeus/Tools/master/hassbian-config/hotfix/install_mosquitto.sh && sudo hassbian-config install mosquitto
```

dpritt, 2018-01-03:
Excellent, that worked a treat.
 
The hassbian-config script does not though, so that needs updating, I
did not try the developers branch but I had run hassbian-config upgrade
hassbian-script before I tried to install Mosquitto.
 
Now to move all my scripts etc. from my old Jessie based Home Assistant
installation.
 
Regards
 
Dave
 
 
>>> On 02/01/2018 at 07:09, in message
<home-assistant/hassbian-scripts/issues/76/354712180@github.com>, Joakim
Sørensen<notifications@github.com> wrote:

sudo curl -o /opt/hassbian/suites/install_mosquitto.sh
https://raw.githubusercontent.com/ludeeus/Tools/master/hassbian-config/hotfix/install_mosquitto.sh
&& sudo hassbian-config install mosquitto

PR #77 (2017-12-18): Mosquitto: Update dependencies in use by the workaround.
Fixes #76
commit 71ce15fd (2017-12-18): Update dependencies in use by the workaround. (#77)

Fixes #76