home-assistant/hassbian-scripts #92: Upgrade script gives wrong discord channel?

DubhAd, 2018-02-04:
As reported in Discord by eFlame, the upgrade script tells people:

_If you have issues with this script, please say something in the #Hassbian channel on Discord_

Except, there is no such channel, and the closest is the channel for Hassbian development, leading to general support questions being asked in the dev channel...

Landrash, 2018-02-04:
@DubhAd Channel name was change to reflect that it's not for general support. Should probably be fixed in the scripts also.

ludeeus, 2018-02-04:
Most scripts I dev now has the new channel name, and the line in question is:
"If you have issues with this script, please say something in the #devs_hassbian channel on Discord." I believe that this is the way to have it, if there is an issue with the scripts that channel would be the correct place other than here to report it.

PR #106 (2018-02-11): Fix wrong Discord channel in scripts
### Description:
Fix wrong Discord channel in scripts.
**Related issue (if applicable):** Fixes #92
closed 2018-02-11 by ludeeus