home-assistant/hassbian-scripts #134: hassbian-config and home assistant disappeared

WagnerBeccard, 2018-03-08:
Trying to run dasher. (its not workig btw^^)
somehow hassbian-config and home assistant disappeared.
but could run home assistand manual.
also systemctl home-assistant@homeassistant.service went missing 

pi@hassbian:~ $ hassbian-config -V
-bash: hassbian-config: command not found

ways i took 
2018-02-09-Hassbian.img
raspy-config
-filesystem & changing time zone
https://github.com/home-assistant/hassbian-scripts/blob/dev/docs/samba.md
https://github.com/danielperna84/hass-configurator#installation
https://github.com/maddox/dasher#raspberry-pi

ludeeus, 2018-03-09:
Sometimes auto remove will remove hassbian-config, and the service file, please reinstall hassbian-config to get it working again:
https://github.com/home-assistant/hassbian-scripts/blob/dev/docs/hassbian_config.md#installation

WagnerBeccard, 2018-03-09:
so maybe apt-mark in the images to protect the config?

DESCRIPTION
       apt-mark can be used as a unified front-end to set various settings for a package, such as marking a package as being automatically/manually
       installed or changing dpkg selections such as hold, install, deinstall and purge which are respected e.g. by apt-get dselect-upgrade or aptitude.

AUTOMATICALLY AND MANUALLY INSTALLED PACKAGES
       When you request that a package is installed, and as a result other packages are installed to satisfy its dependencies, the dependencies are marked
       as being automatically installed, while the package you installed explicitly is marked as manually installed. Once an automatically installed package
       is no longer depended on by any manually installed package it is considered no longer needed and e.g.  apt-get or aptitude will at least suggest
       removing them.

Landrash, 2018-03-09:
Package has been marked since it's incompatible with how it's currently installed. 
If the package is removed you will get a warning from `apt` before removal.

Closing this issue since there's nothing wrong with the image / installation but something that was caused by the user at a later stage.

closed 2018-03-09 by Landrash