home-assistant/hassbian-scripts #141: homebridge.sh

rinho1239104, 2018-03-23:
https://github.com/home-assistant/hassbian-scripts/blob/55751fbd6b5182da56015a3414f3633e8489d0ea/package/opt/hassbian/suites/homebridge.sh
#L57 sudo mkdir /home/homebridge/.homebridge
#L58 sudo touch /home/homebridge/.homebridge/config.json
#L62 cat > /home/homebridge/.homebridge/config.json <<EOF
#L115 echo "path = /home/homebridge/.homebridge" | tee -a /etc/samba/smb.conf

/home/homebridge/.homebridge  -------> /home/pi/.homebridge

ludeeus, 2018-03-23:
@rinho1239104 The service run as `homebridge` https://github.com/home-assistant/hassbian-scripts/blob/master/package/opt/hassbian/suites/homebridge.sh#L96
Therefore `/home/homebridge/.homebridge` should be the correct path, do you have issues with this?

rinho1239104, 2018-03-23:
@ludeeus 
pi@hassbian:~ $ homebridge -D
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
[2018-3-24 00:31:02] Loaded plugin: homebridge-homeassistant
[2018-3-24 00:31:02] Registering platform 'homebridge-homeassistant.HomeAssistant'
[2018-3-24 00:31:02] ---
[2018-3-24 00:31:02] config.json (/home/pi/.homebridge/config.json) not found.

ludeeus, 2018-03-23:
@rinho1239104 That is expected since it's not designed to be run as the `pi` user
Check the documentation for that suite:
https://github.com/home-assistant/hassbian-scripts/blob/dev/docs/homebridge.md

You can start it with `sudo systemctl start homebridge.service`

Landrash, 2018-03-23:
No issue with install script or service. Solution and cause in previous response from @ludeeus

closed 2018-03-23 by Landrash