home-assistant/hassbian-scripts #147: Homebridge script doesn't work for Pi Zero W

cxlwill, 2018-03-25:
**Hassbian-config version (`hassbian-config -V`):**  Latest `v0.8`

**Script/Suite:** `homebridge`

**Description of problem:**  
Homebridge script doesn't work for Pi zero w with Armv6 since `node `8+ could not be installed with default `bash` code: `curl -sL https://deb.nodesource.com/setup_8.x | sudo -E bash -`.

The script would give fake successful feedback after you fully run through whole scripts. If then you type `node -v`, system would fail you with `v4.8.2`. The log in the middle would show the warning: `## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js v4 and later.`

**Expected:**
Since Home Assistant support almost every version of Pi, then maybe we should adjust this script to fit it. 
1. Add Pi model check at the first stage. (Like `uname -a`)
2. Add Pi Zero specific `homebridge`  install script:
```bash
wget -O - https://raw.githubusercontent.com/sdesalas/node-pi-zero/master/install-node-v9.0.0.sh | bash
sudo apt-get install -y npm
```
3. Further adjustment since `homebridge` directory should differ from Pi with Armv7l

Landrash, 2018-03-26:
There's actually multiple scripts and functions that doesn't work with the pi zero.

Currently node is supplied by node source, a fairly reputable source. Moving the install script to one from github isn't something I would be that comfortable with.

I've tagged this and will consider some changes to this in the future.
In the meantime we could add checks to the install scripts to abort if running on a pi zero.

Landrash, 2018-03-27:
Check implemented in #153 . Adding to wishlist and closing issue.

closed 2018-03-27 by Landrash