home-assistant/hassbian-scripts #158: Razberry not turning off bluetooth correctly

leothlon, 2018-04-06:
**Hassbian-config version (`hassbian-config -V`):**  
0.8.1
**Script/Suite:**
Razberry
**Description of problem:**  
Does not seem to turn off bluetooth on raspberry pi 3

it should add "dtoverlay=pi3-miniuart-bt" to bottom of /boot/config.txt file (according to https://www.home-assistant.io/hassio/zwave/)
when i instead added it manually my razberry card started working

**Expected:**
properly add "dtoverlay=pi3-miniuart-bt" to bottom of /boot/config.txt file

**Traceback/log (if applicable):**
```bash

```

**Additional info:**

ludeeus, 2018-04-06:
Can you run this and paste the result here to verify that your unit is in the scope of the script:
`grep Revision /proc/cpuinfo | cut -d: -f2 | tr -d " "`

leothlon, 2018-04-06:
it gives me:

a02082

ludeeus, 2018-04-06:
Thanks, then it's supposed to work :)
Will do some troubleshooting later today :)

PR #159 (2018-04-06): Razberry: Fix for not correctly setting value in /boot/config.txt
## Description:
The check if the value already was present did not work.
This wil correct that.

Also added an validation check for the change.

**Related issue (if applicable):** Fixes #158 

## Checklist (Required):
  - [X] The code change is tested and works locally.
  - [X] The code is compliant with [Contributing guidelines](https://github.com/home-assistant/hassbian-scripts/blob/master/.github/CONTRIBUTING.md)

### If pertinent:
  - [X] Script has validation check of the job.
  - [ ] Created/Updated documentation at `/docs [... cut]
closed 2018-04-06 by Landrash