home-assistant/hassbian-scripts #164: Unable to get Beta version

funnybutrandom, 2018-05-05:
When running hassbian-script 0.9.0 the script is unable to get the beta version of home assistant, it seems to be an issue with getting the current beta version from the json file
[image]

ludeeus, 2018-05-07:
This got broken after pypi changed their domain and structure of the json file.
I'm working on a fix for it.

funnybutrandom, 2018-05-07:
@ludeeus The upgrade script now fails at checking current version:
[image]

PR #165 (2018-05-07): homeassistant: fixes after changes on pypi.
## Description:
Pypi moved their domain and restructured the json file.
This change will make sure that upgrades works agian.

**Related issue (if applicable):** Fixes #164 

## Checklist (Required):
  - [x] The code change is tested and works locally.
  - [x] The code is compliant with [Contributing guidelines](https://github.com/home-assistant/hassbian-scripts/blob/master/.github/CONTRIBUTING.md)

### If pertinent:
  - [x] Script has validation check of the job.
  - [ ] Created/Updated documentation at `/docs`
PR #166 (2018-05-08): homeassistant: Version check for beta
## Description:
When testing the previous PR, I must have been on an older version of HA..
This adds version check to beta script

**Related issue (if applicable):** Fixes #164 

## Checklist (Required):
  - [X] The code change is tested and works locally.
  - [X] The code is compliant with [Contributing guidelines](https://github.com/home-assistant/hassbian-scripts/blob/master/.github/CONTRIBUTING.md)

### If pertinent:
  - [X] Script has validation check of the job.
  - [ ] Created/Updated documentation at `/docs`
closed 2018-05-08 by Landrash