home-assistant/hassbian-scripts #175: AppDaemon install error

osemmler, 2018-07-10:
**Hassbian-config version (`hassbian-config -V`):**  
0.9.1

**Script/Suite:**
AppDaemon

**Description of problem:** 
Error on install AppDaemon suite.

**Traceback/log (if applicable):**
```bash
AppDaemon install script for Hassbian.
Copyright(c) 2017 Fredrik Lindqvist <https://github.com/Landrash>.

Stopping AppDaemon service...
Failed to stop appdaemon@homeassistant.service: Unit appdaemon@homeassistant.service not loaded.
Changing to the homeassistant user...
Changing to AppDaemon venv...
/bin/bash: line 3: /srv/appdaemon/bin/activate: No such file or directory
Installing latest version of AppDaemon...
Collecting wheel
  Using cached https://files.pythonhosted.org/packages/81/30/e935244ca6165187ae8be876b6316ae201b71485538ffac1d718843025a9/wheel-0.31.1-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.31.1
Collecting appdaemon
  Downloading https://files.pythonhosted.org/packages/ec/3c/fb5c9dd92f12d1ff82322de4f9eeab99a32d3ecb6d94e9ccb7823025708d/appdaemon-3.0.1-py3-none-any.whl (5.9MB)
Collecting aiohttp==2.3.10 (from appdaemon)
  Downloading https://www.piwheels.org/simple/aiohttp/aiohttp-2.3.10-cp35-cp35m-linux_armv7l.whl (622kB)
Collecting voluptuous (from appdaemon)
  Using cached https://files.pythonhosted.org/packages/f5/09/a0e20a0bd743131e237128bad3a4f83b283f70c032b7e7c0f06baf7f6862/voluptuous-0.11.1-py2.py3-none-any.whl
Collecting feedparser (from appdaemon)
  Downloading https://www.piwheels.org/simple/feedparser/feedparser-5.2.1-py3-none-any.whl (44kB)
Collecting websocket-client (from appdaemon)
  Downloading https://files.pythonhosted.org/packages/8a/a1/72ef9aa26cfe1a75cee09fc1957e4723add9de098c15719416a1ee89386b/websocket_client-0.48.0-py2.py3-none-any.whl (198kB)
Collecting requests>=2.6.0 (from appdaemon)
  Using cached https://files.pythonhosted.org/packages/65/47/7e02164a2a3db50ed6d8a6ab1d6d60b69c4c3fdf57a284257925dfc12bda/requests-2.19.1-py2.py3-none-any.whl
Collecting iso8601 (from appdaemon)
  Downloading https://files.pythonhosted.org/packages/ef/57/7162609dab394d38bbc7077b7ba0a6f10fb09d8b7701ea56fa1edc0c4345/iso8601-0.1.12-py2.py3-none-any.whl
Collecting pyyaml (from appdaemon)
  HTTP error 404 while getting https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
  Could not install requirement pyyaml from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from appdaemon) because of error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl
Could not install requirement pyyaml from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from appdaemon) because of HTTP error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl for URL https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
Deactivating virtualenv...
/bin/bash: line 11: deactivate: command not found
Starting AppDaemon service...
Failed to start appdaemon@homeassistant.service: Unit appdaemon@homeassistant.service not found.
Checking the installation...

Upgrade done..

To continue have a look at http://appdaemon.readthedocs.io/en/latest/

```

**Additional info:**
First problem i see is that script can't set venv
```bash
Changing to AppDaemon venv...
/bin/bash: line 3: /srv/appdaemon/bin/activate: No such file or directory
```
Second problem is, that script can't install pyyaml:
```bash
Collecting pyyaml (from appdaemon)
  HTTP error 404 while getting https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
  Could not install requirement pyyaml from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from appdaemon) because of error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl
Could not install requirement pyyaml from https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from appdaemon) because of HTTP error 404 Client Error: Not Found for url: https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl for URL https://www.piwheels.org/simple/pyyaml/PyYAML-4.1-cp35-cp35m-linux_armv7l.whl#sha256=a5b125a24244830d574ccfd8fab861198d2ffe491b73179ffd75abf78caee6ff (from https://www.piwheels.org/simple/pyyaml/)
```

ludeeus, 2018-07-10:
It looks like you are trying to use the upgrade script and not the install.
try:
```bash
sudo hassbian-config install appdaemon
```

osemmler, 2018-07-12:
I don't know what I did wrong.
I successfully install AppDaemon manualy.

closed 2018-07-12 by osemmler