home-assistant/hassbian-scripts #194: Install Failed On Raspberry Pi Zero w

madmike89, 2018-10-13:
Make sure your running an up to date version of Hassbian before reporting a problem. If not make sure to specify what version and why so we can recreate the issue in question.

You should only file an issue if you found a bug. Feature and enhancement requests should go in [#33](https://github.com/home-assistant/hassbian-scripts/issues/33) that's dedicated for this usage.

**Hassbian-config version (`hassbian-config -V`):1.21

**Script/Suite:**

**Description of problem:/srv/homeassistant/bin/hass: No such file or directory

**Expected:**

**Traceback/log (if applicable):**
```bash

```

**Additional info:1) Flashed the hassbian 1.21
                          2) Ran sudo hassbian-config install homeassistant --force
                          3) Not working

Logs: sudo journalctl -f -u home-assistant@homeassistant.service
         -- Logs begin at Sat 2017-04-29 17:59:46 SAST. --
Oct 13 15:06:48 hassbian systemd[1]: Starting Home Assistant for homeassistant...
Oct 13 15:06:48 hassbian systemd[1]: Started Home Assistant for homeassistant.
Oct 13 15:06:48 hassbian systemd[1403]: Failed at step EXEC spawning /srv/homeassistant/bin/hass: No such file or directory
Oct 13 15:06:48 hassbian systemd[1]: home-assistant@homeassistant.service: main process exited, code=exited, status=203/EXEC
Oct 13 15:06:48 hassbian systemd[1]: Unit home-assistant@homeassistant.service entered failed state.
Oct 13 15:18:56 hassbian systemd[1]: Starting Home Assistant for homeassistant...
Oct 13 15:18:56 hassbian systemd[1]: Started Home Assistant for homeassistant.
Oct 13 15:18:56 hassbian systemd[1474]: Failed at step EXEC spawning /srv/homeassistant/bin/hass: No such file or directory
Oct 13 15:18:56 hassbian systemd[1]: home-assistant@homeassistant.service: main process exited, code=exited, status=203/EXEC
Oct 13 15:18:56 hassbian systemd[1]: Unit home-assistant@homeassistant.service entered failed state.

sudo systemctl status home-assistant@homeassistant.service
● home-assistant@homeassistant.service - Home Assistant for homeassistant
   Loaded: loaded (/etc/systemd/system/home-assistant@homeassistant.service; enabled)
   Active: failed (Result: exit-code) since Sat 2018-10-13 15:18:56 SAST; 10min ago
  Process: 1474 ExecStart=/srv/homeassistant/bin/hass (code=exited, status=203/EXEC)
 Main PID: 1474 (code=exited, status=203/EXEC)

Oct 13 15:18:56 hassbian systemd[1]: Started Home Assistant for homeassistant.
Oct 13 15:18:56 hassbian systemd[1474]: Failed at step EXEC spawning /srv/ho...y
Oct 13 15:18:56 hassbian systemd[1]: home-assistant@homeassistant.service: ...EC
Oct 13 15:18:56 hassbian systemd[1]: Unit home-assistant@homeassistant.serv...e.
Hint: Some lines were ellipsized, use -l to show in full.

ludeeus, 2018-10-13:
@madmike89 It looks like you have tried a really old version of Hassbian
1.21 are from 30 Apr 2017
Can you try this one <https://github.com/home-assistant/pi-gen/releases/tag/untagged-8e6074a6a6dbf5ed52ca>

madmike89, 2018-10-14:
Hi @ludeeus for some reason the wpa_supplicant.conf file that worked with the previous version i tried does not work.

ludeeus, 2018-10-14:
There were some structural changes to it a while back.
I have used this example with my setup and it has always worked:
<https://www.home-assistant.io/docs/installation/hassbian/installation/#wireless-network>
Make sure you put in the correct country code.

madmike89, 2018-10-14:
Thanks that seems to work. Does this look better.

-- Logs begin at Sat 2018-06-23 20:18:50 UTC. --
Jun 23 20:19:10 hassbian systemd[1]: Started Home Assistant for homeassistant.
Jun 23 20:19:10 hassbian systemd[1]: home-assistant@homeassistant.service: Main process exited, code=exited, status=203/EXEC
Jun 23 20:19:10 hassbian systemd[1]: home-assistant@homeassistant.service: Unit entered failed state.
Jun 23 20:19:10 hassbian systemd[1]: home-assistant@homeassistant.service: Failed with result 'exit-code'.
Oct 14 08:58:27 hassbian systemd[1]: Started Home Assistant for homeassistant.
Oct 14 08:58:27 hassbian systemd[610]: home-assistant@homeassistant.service: Failed at step EXEC spawning /srv/homeassistant/bin/hass: No such file or directory
Oct 14 08:58:27 hassbian systemd[1]: home-assistant@homeassistant.service: Main process exited, code=exited, status=203/EXEC
Oct 14 08:58:27 hassbian systemd[1]: home-assistant@homeassistant.service: Unit entered failed state.
Oct 14 08:58:27 hassbian systemd[1]: home-assistant@homeassistant.service: Failed with result 'exit-code'.

ludeeus, 2018-10-14:
nope :(
It's identical
Try this:
```bash
sudo su -s /bin/bash homeassistant
source /srv/homeassistant/bin/activate
pip install homeassistant
exit
```

madmike89, 2018-10-14:
Its working :):):):):):)