home-assistant/hassbian-scripts #212: hassbian-script 0.10 double entries

JLeandroFreitas, 2018-10-29:
I updated to 0.10 And now have i have double entries.
[image]

ludeeus, 2018-10-29:
@JLeandroFreitas Can you run this command:
```bash
ls /opt/hassbian/suites/
```

JLeandroFreitas, 2018-10-29:
> @JLeandroFreitas Can you run this command:
> 
> ```shell
> ls /opt/hassbian/suites/
> ```

bash
```pi@hassbian:/opt/hassbian/suites $ ls
appdaemon.sh        homeassistant.sh  mosquitto.sh   razberry.sh
cloud9.sh           homebridge.sh     mssql.sh       samba.sh
duckdns.sh          hue.sh            mysql.sh       tradfri.sh
files               libcec.sh         pihole.sh      webterminal.sh
hassbian-script.sh  mariadb.sh        postgresql.sh
hassbian.sh         monitor.sh        python.sh
pi@hassbian:/opt/hassbian/suites $

```

closed 2018-10-29 by JLeandroFreitas