home-assistant/hassbian-scripts #219: Errors updating hassbian-scripts from 0.9.1 to 0.11.0

brianjking, 2018-11-12:
Make sure your running an up to date version of Hassbian before reporting a problem. If not make sure to specify what version and why so we can recreate the issue in question.

You should only file an issue if you found a bug. Feature and enhancement requests should go in [#33](https://github.com/home-assistant/hassbian-scripts/issues/33) that's dedicated for this usage.

**Hassbian-config version (`hassbian-config -V`):**  

```bash
pi@hassbian:~ $ sudo hassbian-config -V
0.9.1
```

**Script/Suite:**

`hassbian-script`

**Description of problem:**  

Attempting to upgrade `hassbian-scripts` package from `0.9.1` to `0.11.0` and receive an error.

**Expected:**

Expect to be able to run `sudo hassbian-config upgrade hassbian-script` and have it download and install the newest general release of the `hassbian-script` package. According to the [releases page](https://github.com/home-assistant/hassbian-scripts/releases) this should be `0.11.0`. 

**Traceback/log (if applicable):**
```bash

pi@hassbian:~ $ sudo hassbian-config upgrade hassbian-script
Hassbian-script upgrade script for Hassbian
Original concept by Ludeeus <https://github.com/ludeeus>

Changing to a temporary folder
Downloading latest release
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2480  100  2480    0     0   9119      0 --:--:-- --:--:-- --:--:--  9151
Installing latest release

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
E: Unsupported file /tmp/hassbian*.deb given on commandline
Cleanup
rm: cannot remove 'hassbian*.deb': No such file or directory

Upgrade is now done.

```

**Additional info:**

* RaspberryPi 3 Model B
* Home Assistant version `0.82.0`
* [My Home Assistant Configuration Files](https://github.com/brianjking/homeassistant-config)

ludeeus, 2018-11-12:
@brianjking Can you try that again now?

brianjking, 2018-11-12:
```bash
pi@hassbian:~ $ sudo hassbian-config -V
0.11.0
```

closed 2018-11-12 by ludeeus