home-assistant/hassbian-scripts #230: Upgrade Home Assistant - wrong

pepeEL, 2019-01-29:
**Hassbian-config version (`hassbian-config -V`):**  
0.11.1
**Script/Suite:**

**Description of problem:**  
From update 0.86.3 when i use script to update HomeAssistant show me after uninstall old version that config is wrong and ask me revert ot no... I always use no and upgrade is ok and i have new version. But show error when use this script.

**Expected:**

**Traceback/log (if applicable):**
```bash

```

**Additional info:**

brianjking, 2019-02-02:
What error did you receive?

ludeeus, 2019-02-03:
The "error" is:

```
Config check failed for new version, do you want to revert? [Y/n] :
```

This is a feature :)

After an update, it does a config check, and if there are issues with your config it allows you to revert to the working version.

pepeEL, 2019-02-04:
But what issue in configuration ? How find issue... because in HA check config show that all is OK

ludeeus, 2019-02-05:
Try running `hashell` then `hacheckconf`
The UI config check does not catch all issues.

pepeEL, 2019-02-07:
pi@hassbian:~ $ hashell
homeassistant@hassbian:/home/pi $ hacheckconf
INFO:homeassistant.util.package:Attempting install of colorlog==4.0.2
Testing configuration at /home/homeassistant/.homeassistant
(homeassistant) homeassistant@hassbian:/home/pi $

pepeEL, 2019-02-21:
Again show this:
``Home Assistant install script for Hassbian.
Copyright(c) 2017 Fredrik Lindqvist <https://github.com/Landrash>.

Checking current version
Setting correct premissions
Changing to the homeassistant user
Changing to Home Assistant venv
Upgrading Home Assistant
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: setuptools in /srv/homeassistant/lib/python3.5/site-packages (40.8.0)
Collecting wheel
  Downloading https://files.pythonhosted.org/packages/96/ba/a4702cbb6a3a485239fbe9525443446203f00771af9ac000fa3ef2788201/wheel-0.33.1-py2.py3-none-any.whl
Installing collected packages: wheel
  Found existing installation: wheel 0.32.3
    Uninstalling wheel-0.32.3:
      Successfully uninstalled wheel-0.32.3
Successfully installed wheel-0.33.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting homeassistant==0.88.0
  Downloading https://files.pythonhosted.org/packages/40/78/07b8b8be760fac280ec4662cf837c9429bcbda446c9ec9eef78cfa4208cb/homeassistant-0.88.0-py3-none-any.whl (3.8MB)
Requirement already satisfied, skipping upgrade: jinja2>=2.10 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (2.10)
Requirement already satisfied, skipping upgrade: aiohttp==3.5.4 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (3.5.4)
Requirement already satisfied, skipping upgrade: PyJWT==1.6.4 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (1.6.4)
Requirement already satisfied, skipping upgrade: pyyaml<4,>=3.13 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (3.13)
Requirement already satisfied, skipping upgrade: voluptuous-serialize==2.0.0 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (2.0.0)
Requirement already satisfied, skipping upgrade: python-slugify==1.2.6 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (1.2.6)
Requirement already satisfied, skipping upgrade: certifi>=2018.04.16 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (2018.11.29)
Requirement already satisfied, skipping upgrade: async-timeout==3.0.1 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (3.0.1)
Requirement already satisfied, skipping upgrade: pip>=8.0.3 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (19.0.1)
Requirement already satisfied, skipping upgrade: bcrypt==3.1.5 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (3.1.5)
Requirement already satisfied, skipping upgrade: attrs==18.2.0 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (18.2.0)
Requirement already satisfied, skipping upgrade: requests==2.21.0 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (2.21.0)
Collecting ruamel.yaml==0.15.88 (from homeassistant==0.88.0)
  Downloading https://www.piwheels.org/simple/ruamel-yaml/ruamel.yaml-0.15.88-cp35-cp35m-linux_armv7l.whl (554kB)
Collecting astral==1.9.2 (from homeassistant==0.88.0)
  Downloading https://files.pythonhosted.org/packages/83/29/d400088aa99e6609ade4d858f5ac7c342d566b4db05ba1ea4c564034625a/astral-1.9.2-py2.py3-none-any.whl
Requirement already satisfied, skipping upgrade: voluptuous==0.11.5 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (0.11.5)
Collecting cryptography==2.5 (from homeassistant==0.88.0)
  Downloading https://www.piwheels.org/simple/cryptography/cryptography-2.5-cp35-cp35m-linux_armv7l.whl (865kB)
Requirement already satisfied, skipping upgrade: pytz>=2018.07 in /srv/homeassistant/lib/python3.5/site-packages (from homeassistant==0.88.0) (2018.9)
Requirement already satisfied, skipping upgrade: MarkupSafe>=0.23 in /srv/homeassistant/lib/python3.5/site-packages (from jinja2>=2.10->homeassistant==0.88.0) (1.1.0)
Requirement already satisfied, skipping upgrade: multidict<5.0,>=4.0 in /srv/homeassistant/lib/python3.5/site-packages (from aiohttp==3.5.4->homeassistant==0.88.0) (4.5.2)
Requirement already satisfied, skipping upgrade: typing-extensions>=3.6.5; python_version < "3.7" in /srv/homeassistant/lib/python3.5/site-packages (from aiohttp==3.5.4->homeassistant==0.88.0) (3.7.2)
Requirement already satisfied, skipping upgrade: chardet<4.0,>=2.0 in /srv/homeassistant/lib/python3.5/site-packages (from aiohttp==3.5.4->homeassistant==0.88.0) (3.0.4)
Requirement already satisfied, skipping upgrade: idna-ssl>=1.0; python_version < "3.7" in /srv/homeassistant/lib/python3.5/site-packages (from aiohttp==3.5.4->homeassistant==0.88.0) (1.1.0)
Requirement already satisfied, skipping upgrade: yarl<2.0,>=1.0 in /srv/homeassistant/lib/python3.5/site-packages (from aiohttp==3.5.4->homeassistant==0.88.0) (1.3.0)
Requirement already satisfied, skipping upgrade: Unidecode>=0.04.16 in /srv/homeassistant/lib/python3.5/site-packages (from python-slugify==1.2.6->homeassistant==0.88.0) (1.0.23)
Requirement already satisfied, skipping upgrade: six>=1.4.1 in /srv/homeassistant/lib/python3.5/site-packages (from bcrypt==3.1.5->homeassistant==0.88.0) (1.12.0)
Requirement already satisfied, skipping upgrade: cffi>=1.1 in /srv/homeassistant/lib/python3.5/site-packages (from bcrypt==3.1.5->homeassistant==0.88.0) (1.11.5)
[... cut ...]
      Successfully uninstalled astral-1.8
  Found existing installation: cryptography 2.3.1
    Uninstalling cryptography-2.3.1:
      Successfully uninstalled cryptography-2.3.1
  Found existing installation: homeassistant 0.87.1
    Uninstalling homeassistant-0.87.1:
      Successfully uninstalled homeassistant-0.87.1
Successfully installed astral-1.9.2 cryptography-2.5 homeassistant-0.88.0 ruamel.yaml-0.15.88
Deactivating virtualenv
Config check failed for new version, do you want to revert? [Y/n] : Restarting Home Assistant
``

ludeeus, 2019-03-30:
Please update to 0.12.0 of `hassbian-config`  before trying this again, that version has more logging that will indicate why the check fails.

Landrash, 2019-05-04:
Closing due to inactivity after function check with dev version.

closed 2019-05-04 by Landrash