home-assistant/hassbian-scripts #250: can i set password for cloud 9

xkinghost, 2019-03-25:
I am using cloud9. it very nice
but anyone can direct use this, can i set  password for cloud 9 UI

ludeeus, 2019-03-30:
Change this argument on line 7 in the service file:
```
-a :
```
to:
```
-a username:password
```

xkinghost, 2019-03-30:
can you help me with detail ?

pi@hassbian:~ $ sudo -u cloud9 -H -s
sudo: unknown user: cloud9
sudo: unable to initialize policy plugin
pi@hassbian:~ $ -a
-bash: -a: command not found
pi@hassbian:~ $ sudo -u homeassistant -H -s
homeassistant@hassbian:/home/pi $ cloud9
bash: cloud9: command not found
homeassistant@hassbian:/home/pi $ source /srv/homeassistant/bin/activate
(homeassistant) homeassistant@hassbian:/home/pi $ cloud9
bash: cloud9: command not found
(homeassistant) homeassistant@hassbian:/home/pi $

ludeeus, 2019-03-30:
The service file is located here  `/etc/systemd/system/cloud9@homeassistant.service`

xkinghost, 2019-03-30:
thanks it worked

PR #251 (2019-04-20): Cloud9: Adds authentication
## Description:

Adds authentication to Cloud9.

**Related issue (if applicable):**  
_Does not fix #250 (existing), but it will for new installations._

## Checklist (Required):
  - [ ] The code change is tested and works locally.
  - [x] The code is compliant with [Contributing guidelines](https://github.com/home-assistant/hassbian-scripts/blob/master/.github/CONTRIBUTING.md)

### If pertinent:
  - [x] Script has validation check of the job.
  - [ ] Created/Updated documentation at `/docs`
closed 2019-03-30 by ludeeus