home-assistant/hassbian-scripts #254: Hassbian base on Debian Buster

Landrash, 2019-04-14:
Issue for tracking the Hassbian release based on Debian Buster.

Landrash, 2019-04-16:
Testing of suites

[] AppDaemon
[] Cloud9
[] custom-component-store
[] Duck DNS
[] Fail2ban
[] Hassbian
[] Home Assistant
[] Homebridge
[] HUE
[] LibCEC
[] Manager
[] MariaDB
[] Monitor
[] Mosquitto
[] Pi-hole
[] PostgreSQL
[] MS SQL
[] RaZberry
[] Samba
[] Trådfri
[] Webterminal

Landrash, 2019-04-16:
Bugg found with local ca-certificates -> https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=923479

Temporary workaround
```
run sudo c_rehash to fix it.
```

Will be added to the install script as a workaround temporarily.

ludeeus, 2019-04-17:
Demn you GitHub 😆

Landrash, 2019-04-17:
My fault. Forgot about linked issues. Should have created a separate issue
for tracking.

Landrash, 2019-05-04:
Testing of suites. New list for reworked version of hassbian-scripts.

- [x] AppDaemon
- [ ] Cloud9
- [ ] custom-component-store
- [ ] Duck DNS
- [ ] Fail2ban
- [ ] Hassbian
- [ ] hassbian-script
- [ ] Home Assistant
- [ ] HUE
- [ ] LibCEC
- [ ] Manager
- [ ] MariaDB  #264
- [ ] Monitor
- [ ] Mosquitto - Fails installation. No repository available for Buster. Fallback to version distributed by Debian? #262 
- [ ] Pi-hole
- [ ] PostgreSQL
- [ ] MS SQL
- [ ] RaZberry
- [ ] Samba 
- [ ] Trådfri
- [ ] Webterminal

Landrash, 2019-05-14:
Package built with #290 
Image being built.
Release package will be uploaded from Auto build.

PR #256 (2019-04-16): Add c_rehash to installation script
## Description:
Runs the `c_rehash` command before installation to verify that local certificates are available.

**Related issue (if applicable):** Fixes #254 (Second comment)

## Checklist (Required):
  - [x] The code change is tested and works locally.
  - [x] The code is compliant with [Contributing guidelines](https://github.com/home-assistant/hassbian-scripts/blob/master/.github/CONTRIBUTING.md)
PR #290 (2019-05-14): Merge Dev into Master
# Description
Merge Dev into Master

Release of new image based on buster

**Related issue (if applicable):** Fixes #254 

## Checklist
- [x] The code change is tested and works locally.
- [x] The code is compliant with [Contributing guidelines][guidelines]
- [x] Updated documentation at `/docs/`

[guidelines]: https://github.com/home-assistant/hassbian-scripts/blob/master/.github/CONTRIBUTING.md
closed 2019-05-14 by Landrash