home-assistant/hassbian-scripts #266: Custom Component Store Wont start

nickm324, 2019-04-20:
**Hassbian-config version (0.12.0`):**  

**Script/Suite:**
Component Store

**Description of problem:**  
custom-component-store@homeassistant.service - custom-component-store for homeassistant
   Loaded: loaded (/etc/systemd/system/custom-component-store@homeassistant.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-04-20 10:19:23 CDT; 3s ago
  Process: 1166 ExecStart=/usr/local/bin/componentstore --port 8120 --nocache --username nick --password $Kyblue88 --ha_path /home/homeassistant/.homeassis
 Main PID: 1166 (code=exited, status=2)

Apr 20 10:19:22 hassbian systemd[1]: Started custom-component-store for homeassistant.
Apr 20 10:19:23 hassbian componentstore[1166]: Usage: componentstore [OPTIONS]
Apr 20 10:19:23 hassbian componentstore[1166]: Try "componentstore --help" for help.
Apr 20 10:19:23 hassbian componentstore[1166]: Error: Got unexpected extra argument (/home/homeassistant/.homeassistant)
Apr 20 10:19:23 hassbian systemd[1]: custom-component-store@homeassistant.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Apr 20 10:19:23 hassbian systemd[1]: custom-component-store@homeassistant.service: Unit entered failed state.
Apr 20 10:19:23 hassbian systemd[1]: custom-component-store@homeassistant.service: Failed with result 'exit-code'.
**Expected:**
it sto start

ludeeus, 2019-04-20:
@nickm324 Can you try putting quotes around your password in the service file?
`/etc/systemd/system/custom-component-store@homeassistant.service`

Then reload and restart
```
sudo systemctl daemon-reload
sudo service custom-component-store@homeassistant restart
```

Landrash, 2019-05-04:
Closing due to inactivity after function check with `dev` version.

closed 2019-05-04 by Landrash