home-assistant/hassbian-scripts #267: Buster and mosquitto

ludeeus, 2019-04-21:
# Description

Currently, there is no official buster repo for mosquitto.

## Possible soulutions:

- Force the usage of the stretch repo #262 
  - This will work currently for both stretch and buster, but will break jessy.
- Build mosquitto from source.
  - This will add complexity, build time and potentially build issues with various arch's.
- Wait for an official buster repo.
  - Who knows when that might happen :/

Landrash, 2019-05-04:
I think the best workaround would be the following.
 - Implement a check for buster.
   - If buster then install skip adding repo and install version included with raspbian/debian buster.

This isn't among your options but it would avoid problems with mixed packages from different distros and potential security problems.

Landrash, 2019-10-26:
https://www.home-assistant.io/blog/2019/10/26/rip-hassbian/

closed 2019-10-26 by Landrash