home-assistant/hassbian-scripts #291: Installation failed Duckdns

batmanday, 2019-05-14:
## Hassbian-config version
0.12.0
  

## Suite name
Fail when i try to install Duck DNS

## Description of problem

```text

```

## Expected result

```text
Do you want to generate certificates to use SSL(https)? [N/y] : y
Changing to homeassistant user...
Cloning into 'dehydrated'...
remote: Enumerating objects: 1961, done.
remote: Total 1961 (delta 0), reused 0 (delta 0), pack-reused 1961
Receiving objects: 100% (1961/1961), 649.51 KiB | 472.00 KiB/s, done.
Resolving deltas: 100% (1232/1232), done.
huyhome.duckdns.org
CHALLENGETYPE='dns-01'
HOOK='./hook.sh'
# INFO: Using main config file /home/homeassistant/dehydrated/config
+ Generating account key...
+ Registering account key with ACME server...
+ Done!
# INFO: Using main config file /home/homeassistant/dehydrated/config
./hook.sh: line 1: 404:: command not found
Creating duckdns folder...
mkdir: cannot create directory ‘duckdns’: File exists
Creating a script file to be used by cron.
Setting permissions...
Creating cron job...
Restarting cron service...
Checking the installation...
find: ‘/home/homeassistant/dehydrated/certs/huyhome.duckdns.org/’: No such file or directory

**Installation failed...**

If you have issues with this script, please say something in the #devs_hassbian channel on Discord.
```

## Traceback/log

```bash

```

**Additional info:**

```text

```

ludeeus, 2019-05-14:
duplicate of https://github.com/home-assistant/hassbian-scripts/issues/288 you will find the solution there,

closed 2019-05-14 by ludeeus

batmanday, 2019-05-14:
thank u so much