home-assistant/hassbian-scripts #294: wpa_supplicant.conf not loading ?

ErikSiemens, 2019-05-16:
Hello,

I downloaded this version as being the last one becourse I had problems with the last one (23-3-2019 I think).
But this version seems not to load wpa_supplicant.conf
With this version I have no internet connection to my RP3+ what so ever

Kind regards

ludeeus, 2019-05-16:
Try the image from 14-04-2019

ErikSiemens, 2019-05-16:
Thats the version where I have this problem with
I am using version 14-4-2019

ludeeus, 2019-05-16:
I have been using Wifi on that without any issues 🤷‍♂ 
Can you post your `wpa_supplicant.conf` file?

ErikSiemens, 2019-05-16:
It's still the same as with ver 23-3-2019

country=NL
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="[SSID]"
    psk="[paswd]"
}

ErikSiemens, 2019-05-17:
I see there is a new release, ver 14-5-2019.
I give this one a try

Landrash, 2019-05-21:
@ErikSiemens Any update?

ErikSiemens, 2019-05-22:
Sorry, I could not find back this post.
After update it seems to work OK
Thanx for you help !

closed 2019-05-22 by ludeeus