home-assistant/hassbian-scripts #295: Version 0.13.0 hassbian-config show not working

seamus65, 2019-05-16:
## Hassbian-config version
hassbian-config: 0.13.0

  

## Suite name
hassbian-config

## Description of problem

```I ran the "hassbian-config upgrade hassbian-script" and it upgraded to 0.13.0, but now hassbian-config show does not work, it tells me to do --help to see available commands.

```

## Expected result

```hassbian-config show should show the list of hassbian-config suites

```

## Traceback/log

```pi@hassbian:~ $ hassbian-config show

  usage: hassbian-config [command] [suite] [options]
  run 'hassbian-config --help' to see all options

```

**Additional info:**

```text

```

Landrash, 2019-05-16:
Confirmed.

Looking into it.

AlbertoLopSie, 2019-08-13:
Bump.

Any fix/workaround on this?

seamus65, 2019-08-14:
I had an operation failure upgrading to 0.96.4 and had to do a fresh install, new sd card, completely clean install then moved over config files, but the same issue still exists with hassbian-config show not working. This proves to me that it wasn't my setup and it is a problem with the code.

Landrash, 2019-10-26:
https://www.home-assistant.io/blog/2019/10/26/rip-hassbian/

closed 2019-10-26 by Landrash