home-assistant/hassbian-scripts #297: Issue tracking for Hassbian based on Debian Buster

Landrash, 2019-05-16:
## Issue tracking for Hassbian based on Debian Buster

This issue is for tracking known problems with Hassbian image based on  Debian Buster.

### Related issues
 * Debian Buster: Mosquitto #267
 * Debian Buster: Pi-Hole #286

HaNgocLong, 2019-05-24:
Problem in release version 2019-05-14. The hass service is not intatiated at the first boot. The wifi connectivity also have some problem even setup the config file as in https://www.home-assistant.io/docs/installation/hassbian/installation/

Landrash, 2019-05-24:
Hass service isn't enabled until the installer has run it's course. (30 min on a base install)

Wifi problem is an upstream problem that I don't have a simple solution for currently.

SeTeM, 2019-05-30:
Missing release of deCONZ: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1558

Landrash, 2019-05-30:
@SeTeM Deconz-rest-plugin don't have a release for buster it seems. This isn't related to the Hassbian image and nothing we can or will correct.

LeonDragon, 2019-05-31:
There is a conflict when using zigbee2mqtt in this version. In details, the version of node.js needs to be downgraded to v10.x  to compatible with this component. the npm commands seem to have some error?

flex-0, 2019-06-14:
Deconz is not working with buster by now. Is there a workaround for now or do we have to wait until Deconz releases a new version?

Landrash, 2019-06-15:
I've never used deconz so I think that's something you'll have to ask
Deconz about.

Hassbian is just a generic Debian Buster under it all.

Landrash, 2019-10-26:
https://www.home-assistant.io/blog/2019/10/26/rip-hassbian/

closed 2019-10-26 by Landrash