home-assistant/hassbian-scripts #304: Python 3.7 sqlite issue

ImNightwing, 2019-06-10:
## Hassbian-config version

0.8.0

## Suite name

python

## Description of problem

```
after using upgrade python, my home assistant can no longer load sqlite3. Home assistant doesn't load up correctly with a default database giving me an error:
ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: No module named '_sqlite3' (retrying in 3 seconds)

```

## Expected result

```
For sqlite to be configured correctly with python 3.7.2
```

## Traceback/log

```bash

```

**Additional info:**

```text

```

Landrash, 2019-06-11:
I'm unable to reproduce this error.

damyanor, 2019-06-11:
I'm facing the same issue. As a result recorder, history and z-wave components do not work. 

> 2019-06-11 14:54:30 ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: No module named '_sqlite3' (retrying in 3 seconds)
> 2019-06-11 14:54:30 ERROR (MainThread) [homeassistant.setup] Setup failed for recorder: Component failed to initialize.
> 2019-06-11 14:54:32 WARNING (MainThread) [openzwave] pysqlite is not installed
> 2019-06-11 14:54:33 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Z-Wave (import from configuration.yaml) for zwave
> Traceback (most recent call last):
>   File "/srv/homeassistant/lib/python3.7/site-packages/openzwave/network.py", line 324, in __init__
>     self.dbcon = lite.connect(os.path.join(self._options.user_path, 'pyozw.sqlite'), check_same_thread=False)
> NameError: name 'lite' is not defined
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/config_entries.py", line 273, in async_setup
>     hass, self)
>   File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/zwave/__init__.py", line 290, in async_setup_entry
>     network = hass.data[DATA_NETWORK] = ZWaveNetwork(options, autostart=False)
>   File "/srv/homeassistant/lib/python3.7/site-packages/openzwave/network.py", line 329, in __init__
>     except lite.Error as e:
> NameError: name 'lite' is not defined
> 2019-06-11 14:54:33 ERROR (MainThread) [homeassistant.setup] Unable to set up dependencies of history. Setup failed for dependencies: recorder
> 2019-06-11 14:54:33 ERROR (MainThread) [homeassistant.setup] Setup failed for history: Could not set up all dependencies.

cunninr2, 2019-07-05:
I have the same problem. It appears the Hassbian config upgrade python script compiles a new python instance without the libraries for sqllite3 (which I think is an optional compile).   I think the compile in this script should have been "./configure --enable-shared --enable-loadable-sqlite-extensions". Also I cannot re-run the script because it now says the latest version of python is installed. So pretty stuck here.

Landrash, 2019-07-08:
Is this by any chance a Raspian Stretch problem your having?

It that's the case I would recommend upgrading to the Buster based image.

Landrash, 2019-10-26:
https://www.home-assistant.io/blog/2019/10/26/rip-hassbian/

closed 2019-10-26 by Landrash