home-assistant/hassbian-scripts #305: Cannot upgrade HA from 0.94 to 0.95

DominikPalo, 2019-07-05:
## Hassbian-config version
0.13.1

## Suite name
hassbian

## Description of problem
I'm trying to upgrade my HA to the latest version (0.95.4), but after running the command `sudo hassbian-config upgrade homeassistant` I still have the old version (0.94.0). I waited for more than an hour, restarted my RPi, but there is still version 0.94.0 running. I was able to upgrade my HA with that command many times without issues before.

## Expected result
HA is upgraded to the latest version

## Traceback/log

```bash
Generating system information...

  Home Assistant:        0.94.0
  Home Assistant Python: 3.7.3
  hassbian-config:       0.13.1

  Linux hassbian 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux

Suite:     Home Assistant
Installed: true
Contributed to the project by Lindqvist <https://github.com/Landrash>
Installs the base homeassistant package onto this system.

Checking current version
Setting correct permissions
chown: cannot access '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/verisure/__pycache__': Structure needs cleaning
chown: cannot access '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/oem': Structure needs cleaning
chown: cannot access '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/harmony': Structure needs cleaning
chown: cannot access '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/bt_home_hub_5/__pycache__': Structure needs cleaning
chown: changing ownership of '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/bt_home_hub_5/manifest.json': Structure needs cleaning
chown: changing ownership of '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/bt_home_hub_5/device_tracker.py': Structure needs cleaning
chown: changing ownership of '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/swisscom/__init__.py': Structure needs cleaning
chown: changing ownership of '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/swisscom/__pycache__/device_tracker.cpython-37.pyc': Structure needs cleaning
chown: cannot access '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/eight_sleep': Structure needs cleaning
chown: cannot access '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/arduino': Structure needs cleaning
chown: cannot access '/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/flock': Structure needs cleaning
Upgrading Home Assistant
Changing to homeassistant user...
Changing to Home Assistant venv...
Updating Python dependencies...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: pip in /srv/homeassistant/lib/python3.7/site-packages (19.1.1)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: setuptools in /srv/homeassistant/lib/python3.7/site-packages (41.0.1)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: wheel in /srv/homeassistant/lib/python3.7/site-packages (0.33.4)
Installing Python packages...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: homeassistant==0.95.4 in /srv/homeassistant/lib/python3.7/site-packages (0.95.4)
Requirement already satisfied, skipping upgrade: pyyaml==5.1 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (5.1)
Requirement already satisfied, skipping upgrade: requests==2.22.0 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (2.22.0)
Requirement already satisfied, skipping upgrade: python-slugify==3.0.2 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (3.0.2)
Requirement already satisfied, skipping upgrade: importlib-metadata==0.15 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (0.15)
Requirement already satisfied, skipping upgrade: cryptography==2.6.1 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (2.6.1)
Requirement already satisfied, skipping upgrade: pytz>=2019.01 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (2019.1)
Requirement already satisfied, skipping upgrade: voluptuous-serialize==2.1.0 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (2.1.0)
Requirement already satisfied, skipping upgrade: aiohttp==3.5.4 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (3.5.4)
Requirement already satisfied, skipping upgrade: jinja2>=2.10 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (2.10.1)
Requirement already satisfied, skipping upgrade: pip>=8.0.3 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (19.1.1)
Requirement already satisfied, skipping upgrade: async-timeout==3.0.1 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (3.0.1)
Requirement already satisfied, skipping upgrade: bcrypt==3.1.6 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (3.1.6)
Requirement already satisfied, skipping upgrade: astral==1.10.1 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (1.10.1)
Requirement already satisfied, skipping upgrade: PyJWT==1.7.1 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (1.7.1)
Requirement already satisfied, skipping upgrade: voluptuous==0.11.5 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (0.11.5)
Requirement already satisfied, skipping upgrade: ruamel.yaml==0.15.97 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (0.15.97)
Requirement already satisfied, skipping upgrade: attrs==19.1.0 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (19.1.0)
Requirement already satisfied, skipping upgrade: certifi>=2018.04.16 in /srv/homeassistant/lib/python3.7/site-packages (from homeassistant==0.95.4) (2019.3.9)
Requirement already satisfied, skipping upgrade: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /srv/homeassistant/lib/python3.7/site-packages (from requests==2.22.0->homeassistant==0.95.4) (1.25.3)
Requirement already satisfied, skipping upgrade: idna<2.9,>=2.5 in /srv/homeassistant/lib/python3.7/site-packages (from requests==2.22.0->homeassistant==0.95.4) (2.8)
Requirement already satisfied, skipping upgrade: chardet<3.1.0,>=3.0.2 in /srv/homeassistant/lib/python3.7/site-packages (from requests==2.22.0->homeassistant==0.95.4) (3.0.4)
Requirement already satisfied, skipping upgrade: text-unidecode==1.2 in /srv/homeassistant/lib/python3.7/site-packages (from python-slugify==3.0.2->homeassistant==0.95.4) (1.2)
Requirement already satisfied, skipping upgrade: zipp>=0.5 in /srv/homeassistant/lib/python3.7/site-packages (from importlib-metadata==0.15->homeassistant==0.95.4) (0.5.1)
Requirement already satisfied, skipping upgrade: asn1crypto>=0.21.0 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography==2.6.1->homeassistant==0.95.4) (0.24.0)
Requirement already satisfied, skipping upgrade: six>=1.4.1 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography==2.6.1->homeassistant==0.95.4) (1.12.0)
Requirement already satisfied, skipping upgrade: cffi!=1.11.3,>=1.8 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography==2.6.1->homeassistant==0.95.4) (1.12.3)
Requirement already satisfied, skipping upgrade: yarl<2.0,>=1.0 in /srv/homeassistant/lib/python3.7/site-packages (from aiohttp==3.5.4->homeassistant==0.95.4) (1.3.0)
Requirement already satisfied, skipping upgrade: multidict<5.0,>=4.0 in /srv/homeassistant/lib/python3.7/site-packages (from aiohttp==3.5.4->homeassistant==0.95.4) (4.5.2)
Requirement already satisfied, skipping upgrade: MarkupSafe>=0.23 in /srv/homeassistant/lib/python3.7/site-packages (from jinja2>=2.10->homeassistant==0.95.4) (1.1.1)
Requirement already satisfied, skipping upgrade: pycparser in /srv/homeassistant/lib/python3.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography==2.6.1->homeassistant==0.95.4) (2.19)
Deactivating virtualenv...
Restarting Home Assistant
Checking the installation...

Operation completed...

Note that it may take some time to start up after an upgrade.
```

ludeeus, 2019-07-05:
From what I can tell the script has done what it's supposed to do.

`Requirement already up-to-date: homeassistant==0.95.4`

This says that you have it installed already, try clearing your cache if it's the /dev-info page you use to see the old version.

If you still need support with this visit the #hassbian channel on discord.
https://discord.gg/c5DvZ4e

closed 2019-07-05 by ludeeus

DominikPalo, 2019-07-05:
Clearing the browser cache didn't help - also as you can see on the first line of my output (the output is from the second run of the upgrade script):
` Home Assistant:        0.94.0` - so it still running the 0.94.0 version