home-assistant/hassbian-scripts #306: sudo hassbian-config install samba fails

baegopooh, 2019-07-17:
## Hassbian-config version

  
0.13.1
## Suite name

samba
## Description of problem

```
 sudo hassbian-config install samba

cmd fails.

```

## Expected result

```

successful installation of samba on hassbian

```

## Traceback/log

```

Generating system information...

  Home Assistant:        0.95.4
  Home Assistant Python: 3.7.3
  hassbian-config:       0.13.1

  Linux hassbian 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l GNU/Linux

Suite:     Samba
Installed: false
Contributed to the project by Landrash <https://github.com/Landrash>
Installs the samba package for sharing the Hassbian configuration files, over the network.

Running apt-get preparation
Updating apt...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Ign:2 https://gitlab.com/hassbian/repository/raw/master buster InRelease
Get:3 https://gitlab.com/hassbian/repository/raw/master buster Release [3,110 B]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [205 kB]
Ign:5 https://gitlab.com/hassbian/repository/raw/master buster Release.gpg
Get:6 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Reading package lists...
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
Installing packages...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  libdrm-amdgpu1 libllvm7
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  attr ibverbs-providers libavahi-client3 libboost-atomic1.67.0
  libboost-iostreams1.67.0 libboost-regex1.67.0 libboost-system1.67.0
  libboost-thread1.67.0 libcephfs2 libcups2 libgfapi0 libgfrpc0 libgfxdr0
  libglusterfs0 libibverbs1 libjansson4 libldb1 librados2 libtdb1 libtevent0
  python-crypto python-dnspython python-gpg python-ldb python-samba
  python-talloc python-tdb samba-common samba-common-bin samba-dsdb-modules
  samba-libs samba-vfs-modules tdb-tools
Suggested packages:
  cups-common python-crypto-doc bind9 bind9utils ctdb ldb-tools ntp | chrony
  smbldap-tools ufw winbind heimdal-clients
The following NEW packages will be installed:
  attr ibverbs-providers libavahi-client3 libboost-atomic1.67.0
  libboost-iostreams1.67.0 libboost-regex1.67.0 libboost-system1.67.0
  libboost-thread1.67.0 libcephfs2 libcups2 libgfapi0 libgfrpc0 libgfxdr0
  libglusterfs0 libibverbs1 libjansson4 libldb1 librados2 libtdb1 libtevent0
  python-crypto python-dnspython python-gpg python-ldb python-samba
  python-talloc python-tdb samba samba-common samba-common-bin
  samba-dsdb-modules samba-libs samba-vfs-modules tdb-tools
0 upgraded, 34 newly installed, 0 to remove and 7 not upgraded.
Need to get 8,979 kB/24.7 MB of archives.
After this operation, 94.8 MB of additional disk space will be used.
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf samba-libs armhf 2:4.9.5+dfsg-3
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf python-samba armhf 2:4.9.5+dfsg-3
  404  Not Found [IP: 93.93.128.193 80]
Err:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf samba-common all 2:4.9.5+dfsg-3
  404  Not Found [IP: 93.93.128.193 80]
Err:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf samba-common-bin armhf 2:4.9.5+dfsg-3
  404  Not Found [IP: 93.93.128.193 80]
Err:5 http://raspbian.raspberrypi.org/raspbian buster/main armhf samba armhf 2:4.9.5+dfsg-3
  404  Not Found [IP: 93.93.128.193 80]
Err:6 http://raspbian.raspberrypi.org/raspbian buster/main armhf samba-dsdb-modules armhf 2:4.9.5+dfsg-3
  404  Not Found [IP: 93.93.128.193 80]
Err:7 http://raspbian.raspberrypi.org/raspbian buster/main armhf samba-vfs-modules armhf 2:4.9.5+dfsg-3
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-libs_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/python-samba_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-common_4.9.5+dfsg-3_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-common-bin_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-dsdb-modules_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-vfs-modules_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Adding homeassistant Samba user
/opt/hassbian/suites/samba/install: line 14: smbpasswd: command not found
Adding shared folder for Home Assistant configuration directory
tee: /etc/samba/smb.conf: No such file or directory
[homeassistant]
path = /home/homeassistant/.homeassistant
writeable = yes
guest ok = yes
create mask = 0644
directory mask = 0755
force user = homeassistantRestarting Samba service
Failed to restart smbd.service: Unit smbd.service not found.
Checking the installation...

Operation failed...

If you have issues with this script, please say something in the #hassbian channel on Discord.

```

**Additional info:**

```using uptodate hassbian image from HA homepage. did upgrade hassbian and HA.

```

ludeeus, 2019-07-17:
Try to run
`sudo hassbian-config upgrade hassbian`
then try again.

baegopooh, 2019-07-17:
I run 

sudo hassbian-config upgrade hassbian

~~~~

Operation completed...

Note that you may need to reboot the raspberry Pi for some updates to take effect.
pi@hassbian:~ $ sudo reboot
Using username "pi".
pi@192.168.2.57's password:
Send automatic password
 _                   _     _
| |__   __ _ ___ ___| |__ (_) __ _ _ __
| '_ \ / _` / __/ __| '_ \| |/ _` | '_ \
| | | | (_| \__ \__ \ |_) | | (_| | | | |
|_| |_|\__,_|___/___/_.__/|_|\__,_|_| |_|

system info:
  Distro......: Raspbian GNU/Linux buster/sid
  Kernel......: Linux 4.19.57-v7+

  Uptime......: up 0 minutes
  Load........: 0.67 (1m), 0.17 (5m), 0.06 (15m)
  Processes...: 118 (root), 8 (user) | 126 (total)

  CPU.........: ARMv7 Processor rev 4 (v7l)
  Memory......: 93Mi used, 721Mi free, 926Mi in total

services:
  home-assistant@homeassistant:  active
Last login: Wed Jul 17 13:50:35 2019 from 192.168.2.199
pi@hassbian:~ $ sudo hassbian-config install samba
Generating system information...

  Home Assistant:        0.95.4
  Home Assistant Python: 3.7.3
  hassbian-config:       0.13.1

  Linux hassbian 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

Suite:     Samba
Installed: false
Contributed to the project by Landrash <https://github.com/Landrash>
Installs the samba package for sharing the Hassbian configuration files, over the network.

Running apt-get preparation
Updating apt...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Ign:2 https://gitlab.com/hassbian/repository/raw/master buster InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:4 https://gitlab.com/hassbian/repository/raw/master buster Release [3,110 B]
Ign:5 https://gitlab.com/hassbian/repository/raw/master buster Release.gpg
Reading package lists...
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
Installing packages...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  libdrm-amdgpu1 libllvm7
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
[... 35 lines cut ...]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-libs_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/python-samba_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-common_4.9.5+dfsg-3_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-common-bin_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-dsdb-modules_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-vfs-modules_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Adding homeassistant Samba user
/opt/hassbian/suites/samba/install: line 14: smbpasswd: command not found
Adding shared folder for Home Assistant configuration directory
tee: /etc/samba/smb.conf: No such file or directory
[homeassistant]
path = /home/homeassistant/.homeassistant
writeable = yes
guest ok = yes
create mask = 0644
directory mask = 0755
force user = homeassistantRestarting Samba service
Failed to restart smbd.service: Unit smbd.service not found.
Checking the installation...

Operation failed...

If you have issues with this script, please say something in the #hassbian channel on Discord.

ludeeus, 2019-07-17:
How "fun", I'll be home in a few hours, can test on my HW then.

baegopooh, 2019-07-17:
Thanks. it looks like there's something wrong with version numbers. but don't know how to fix it.

ludeeus, 2019-07-17:
" 404  Not Found [IP: 93.93.128.193 80]" is the issue ;)
But not sure how to fix it yet :)

baegopooh, 2019-07-17:
Yep. for the last line
"
E: Failed to fetch  http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/samba-vfs-modules_4.9.5+dfsg-3_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
"
in this directory(http://raspbian.raspberrypi.org/raspbian/pool/main/s/samba/)

there is 3 files

  | samba-vfs-modules_4.2.14+dfsg-0+deb8u13_armhf.deb | 23-May-2019 10:08 | 299K |   | Details
-- | -- | -- | -- | -- | --
  | samba-vfs-modules_4.5.16+dfsg-1+deb9u1_armhf.deb | 08-Apr-2019 16:08 | 327K |   | Details
  | samba-vfs-modules_4.9.5+dfsg-5_armhf.deb | 22-Jun-2019 10:08 | 391K |   | Details

..so there's no 'samba-vfs-modules_4.9.5+dfsg-3_armhf.deb'. there's only ' samba-vfs-modules_4.9.5+dfsg-5_armhf.deb'

ludeeus, 2019-07-17:
Correct :)

Landrash, 2019-07-19:
Looked into this and it seem the server your getting as your upstream for Raspbian(taken out of a pool) has incorrect packages. 
I'm unable to replicate this in Sweden with the servers I'm getting but it should be addressed as soon as the server your using is updated.

bigbigblue, 2019-08-04:
I have exactly the same problem today. Is there any way to override the server so I can request the install from a working server? 16 days since the previous comment and the server has not been updated - seems rather a long time......

Landrash, 2019-08-04:
Have a look at how to specify a server for Raspbian since that's the base
of Hassbian.

baegopooh, 2019-08-11:
Don't know exactly why. but as i got some error msg with apt-get about 

> E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'

I ran
` sudo apt update --allow-releaseinfo-change`
 `sudo apt-get upgrade`
`sudo apt-get update`
 
And now problem solved! 

`force user = homeassistantRestarting Samba service
Checking the installation...

Operation completed...

Configuration is now available as a Samba share at \\192.168.2.57\homeassistant
`

bigbigblue, 2019-08-11:
> Don't know exactly why. but as i got some error msg with apt-get about
>
>
> I ran
> ` sudo apt update --allow-releaseinfo-change`
> `sudo apt-get upgrade`
> `sudo apt-get update`
>
> And now problem solved! [...]

Thanks I ran the above commands, then 'sudo hassbian-config install samba' and samba installed OK.
Thanks

Landrash, 2019-08-14:
Solved in comments :)

closed 2019-08-14 by Landrash