home-assistant/hassbian-scripts #307: OPENSSL_1_1_1

joseant, 2019-07-18:
## Hassbian-config version
0.13.1

## Suite name
hassbian
## Description of problem

I got this error when I run home assistant 0.96
ImportError: /usr/lib/arm-linux-gnueabihf/libssl.so.1.1: version `OPENSSL_1_1_1' not found (required by /srv/homeassistant/lib/python3.7/site-packages/cryptography/hazmat/bindings/_openssl.abi3.so)

I think it is the old version openssl, any idea how update it?

thanks.

DubhAd, 2019-07-18:
This relates to the use of PiWheels, as being discussed [over here](https://github.com/home-assistant/home-assistant/issues/25248), when you're on a Pi using a version of Raspbian before Buster, and Python 3.7.

The answer is to disable PiWheels, and then reinstall `cryptography`:

1) Edit `/etc/pip.conf` and comment out `extra-index-url=https://www.piwheels.org/simple` by placing a `#` at the start of the line
2) Activate the venv, and as the Home Assistant user uninstall then reinstall `cryptography`:

```
$ sudo -u homeassistant -H -s
$ source /srv/homeassistant/bin/activate
$ pip uninstall cryptography
$ pip install cryptography
```

(NB: tested on a non-Hassbian install)

gadgetchnnel, 2019-07-18:
> This relates to the use of PiWheels, as being discussed [over here](https://github.com/home-assistant/home-assistant/issues/25248), when you're on a Pi using a version of Raspbian before Buster, and [...]

I've tried this but I get the following error (on a regular virtual environment installation on Raspbian Stretch):

 ```
build/temp.linux-armv7l-3.7/_openssl.c:498:30: fatal error: openssl/opensslv.h: No such file or directory
   #include <openssl/opensslv.h>
                                ^
  compilation terminated.
  error: command 'gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
  Running setup.py clean for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
```

joseant, 2019-07-18:
> This relates to the use of PiWheels, as being discussed [over here](https://github.com/home-assistant/home-assistant/issues/25248), when you're on a Pi using a version of Raspbian before Buster, and [...]

I have try that, but it allways install the same version of cryptography the 2.7

[image]

gadgetchnnel, 2019-07-18:
> I have try that, but it allways install the same version of cryptography the 2.7

It will still install version 2.7, but it should install it referencing the correct version of OpenSSL.
In the end to get it working I needed to follow the steps as above but run ```sudo apt-get install --reinstall libssl-dev``` before re-installing cryptography.

joseant, 2019-07-18:
I tried that too, but there has been no luck

[image]

> It will still install version 2.7, but it should install it referencing the correct version of OpenSSL.
> In the end to get it working I needed to follow the steps as above but run `sudo apt-get [...]

gadgetchnnel, 2019-07-18:
> I tried that too, but there has been no luck
Did you definitely edit `/etc/pip.conf` and comment out the `extra-index-url=https://www.piwheels.org/simple` line? If you did, the output from ```pip install cryptography``` should have more than that, for me it was like:

```
Collecting cryptography
  Using cached https://files.pythonhosted.org/packages/c2/95/f43d02315f4ec074219c6e3124a87eba1d2d12196c2767fadfdc07a83884/cryptography-2.7.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography) (1.12.3)
Requirement already satisfied: asn1crypto>=0.21.0 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography) (0.24.0)
Requirement already satisfied: six>=1.4.1 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography) (1.12.0)
Requirement already satisfied: pycparser in /srv/homeassistant/lib/python3.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography) (2.19)
Building wheels for collected packages: cryptography
  Building wheel for cryptography (PEP 517) ... done
  Stored in directory: /home/homeassistant/.cache/pip/wheels/d0/02/96/64b1439e5409591b6b0294d1da2f66a4ae4f0548d1bdb225b7
Successfully built cryptography
```
The output you are seeing looks like it is still using PiWheels.

joseant, 2019-07-18:
I had comented the line in the `etc/pip.conf`

I have installed it with pip install cryptography --no-cache-dir  to avoid cache, and now the output is similar yours, but still not working

```
homeassistant@hassbian:/home/pi $ source /srv/homeassistant/bin/activate
(homeassistant) homeassistant@hassbian:/home/pi $ pip uninstall cryptography
Uninstalling cryptography-2.7:
  Would remove:
    /srv/homeassistant/lib/python3.7/site-packages/cryptography-2.7.dist-info/*
    /srv/homeassistant/lib/python3.7/site-packages/cryptography/*
Proceed (y/n)? y
  Successfully uninstalled cryptography-2.7
(homeassistant) homeassistant@hassbian:/home/pi $ pip install cryptography --no-cache-dir
Collecting cryptography
  Downloading https://files.pythonhosted.org/packages/c2/95/f43d02315f4ec074219c6e3124a87eba1d2d12196c2767fadfdc07a83884/cryptography-2.7.tar.gz (495kB)
     |████████████████████████████████| 501kB 1.9MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: six>=1.4.1 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography) (1.12.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography) (1.12.3)
Requirement already satisfied: asn1crypto>=0.21.0 in /srv/homeassistant/lib/python3.7/site-packages (from cryptography) (0.24.0)
Requirement already satisfied: pycparser in /srv/homeassistant/lib/python3.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography) (2.19)
Building wheels for collected packages: cryptography
  Building wheel for cryptography (PEP 517) ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-b377mwda/wheels/d0/02/96/64b1439e5409591b6b0294d1da2f66a4ae4f0548d1bdb225b7
Successfully built cryptography
Installing collected packages: cryptography
Successfully installed cryptography-2.7
(homeassistant) homeassistant@hassbian:/home/pi $ openssl
openssl: /usr/lib/arm-linux-gnueabihf/libssl.so.1.1: version `OPENSSL_1_1_1' not found (required by openssl)
openssl: /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1: version `OPENSSL_1_1_1' not found (required by openssl)
(homeassistant) homeassistant@hassbian:/home/pi $ 

```

gadgetchnnel, 2019-07-18:
@joseant What is the result if you enter the following command
```which openssl```
The result should be
```/usr/bin/openssl```

If it is different, you may have tried to manually install OpenSSL 1.1.1 from source and it is trying to use this, rather than version 1.1.0. I did try doing that at one point, and then had to go into the directory I had installed it from and type ```sudo make uninstall``` to remove it (you may also need to run ```sudo apt-get install openssl``` after doing this to make sure you get OpenSSL 1.1.0 back).

frenck, 2019-07-18:
The reason why this broke, PiWheels builds wheels against Debian Buster, which has a newer glibc, which now causes issues if you are still on Debian Stretch.

There are 2 possible option to get this fixed:

### Option 1: _IMHO the best / future proof /preferred option_
Upgrade your distribution to Debian Buster. After upgrade, everything works as intended.

Pros: more future proof, Piwheels work.
Cons: Upgrades your whole system and might affect other things you might be running on your device.

Steps:
- Edit `stretch` to `buster` in the following files:
    * `/etc/apt/sources.list`
    * `/etc/apt/sources.list.d/raspi.list`
    * `/etc/apt/sources.list.d/hassbian.list`
- `sudo apt-get upgrade`
- `sudo apt-get -y dist-upgrade`
- `sudo reboot`

### Option 2:
Disable Piwheels prebuild packages.

Pros: Quick and easy.
Cons: You'll end up with a possible slow updating/upgrading/installation when new packages need to be installed, since it needs to compile them from source. You are still on Stretch, which is not really future proof.

Steps:
- Edit `/etc/pip.conf` and comment out the `extra-index-url` like this:

    ```ini
    [global]
    # extra-index-url=https://www.piwheels.org/simple
   ```

- `pip uninstall cryptography`
- `pip install --no-cache cryptography==2.7`

Ernst79, 2019-07-18:
Thanks for these steps to solve the issue. I think the second step in option 1 should be `sudo apt-get update`

frenck, 2019-07-18:
@Ernst79 Adjusted, thanks!

frenck, 2019-07-18:
Closing up this issue, since there is nothing left to do. See the answer above for a solution.

closed 2019-07-18 by frenck

joseant, 2019-07-18:
thanks, option 1 worked for me

minsuke, 2019-07-19:
thx frenck, but I think I should reinstall home assistant sooner or late. : )

soccerob, 2019-07-22:
I was able to move from stretch to buster, but I was using my synology NAS to host a MariaDB database for my recorder, and now I'm getting the following error:
`ERROR (Recorder) [homeassistant.components.recorder] Error during connection setup: libmariadbclient.so.18: cannot open shared object file: No such file or directory (retrying in 3 seconds)`

I tried running sudo apt-get install libmysqlclient but I receive an error that there is no installation candidate.