home-assistant/hassbian-scripts #313: wrong glibc version after update

m0rph13, 2019-08-03:
## Hassbian-config version

0.13.1

## Suite name

?

## Description of problem

```
After running the script 
sudo hassbian-config upgrade python
to upgrade Python to the lastest stable version, Home assistant does not start.
```

## Expected result

```
Home assistant should start successfully
```

## Traceback/log

```bash

2019-08-03 12:12:35 WARNING (Recorder) [homeassistant.components.recorder] Ended unfinished session (id=16 from 2019-08-03 10:10:13.284521)
2019-08-03 12:12:36 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Z-Wave for zwave
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/config_entries.py", line 153, in async_setup
    hass, self)
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/zwave/__init__.py", line 265, in async_setup_entry
    from openzwave.option import ZWaveOption
  File "/srv/homeassistant/lib/python3.7/site-packages/openzwave/option.py", line 29, in <module>
    import libopenzwave
ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /srv/homeassistant/lib/python3.7/site-packages/libopenzwave.cpython-37m-arm-linux-gnueabihf.so)
```

hiwok, 2019-09-10:
Hello,

Same problem here. Please find my solution :

Verify GLIBC_2.28 is not present : 
strings /lib/arm-linux-gnueabihf/libc.so.6 | grep GLIBC

Add testing in your sources :
"deb http://http.debian.net/debian/ testing main contrib non-free" >> /etc/apt/sources.list

Upgrade your system (Care, it's a testing package)!
sudo apt-get update
sudo apt-cache search lib6c
sudo apt-get -f install libc6

strings /lib/arm-linux-gnueabihf/libc.so.6 | grep GLIBC_2.28

It's now present.

kwende, 2019-09-14:
Running 

`"deb http://http.debian.net/debian/ testing main contrib non-free" >> /etc/apt/sources.list`

Results in:
`bash: deb http://http.debian.net/debian/ testing main contrib non-free: No such file or directory`

Nocsaron, 2019-09-19:
The command should be
 `echo "deb http://http.debian.net/debian/ testing main contrib non-free" >> /etc/apt/sources.list`

You can also manually add the line in the file using vim or your favorite text editor

piet206, 2019-09-26:
Thanks, that worked for me.

stephanvierkant, 2019-09-29:
W: GPG error: http://cdn-fastly.deb.debian.org/debian testing InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 04EE7237B7D453EC NO_PUBKEY 648ACFD622F3D138
W: The repository 'http://http.debian.net/debian testing InRelease' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.

I've ran into the same problem after updating Homeassistant.

Landrash, 2019-10-26:
https://www.home-assistant.io/blog/2019/10/26/rip-hassbian/

closed 2019-10-26 by Landrash