home-assistant/hassbian-scripts #317: problem on version update

platini76, 2019-09-05:
after update from 0.96.3 to 0.96.5 and again to 0.97 I have wrong version displayed

someone know how to solve?

[image]
[image]

ludeeus, 2019-09-05:
You probably have 2 instances of Home Assistant installed.
Please use our forum, or Discord for support :)

https://www.home-assistant.io/help/#communication-channels

closed 2019-09-05 by ludeeus

platini76, 2019-09-05:
I don't know if I have two instances but... I use hassbian-config update for 1 year without issue... then 1 month ago it happens... I  think this is a problem of the script...
I tried to use the forum but they say don't use this script...
https://community.home-assistant.io/t/homeassistant-version-wrong-displayed/130667